microsoft/vscode-arduino #975: No way to pass extra compile flags and defines

SlavDimov, 2020-02-29:
I need to pass defines to the compiler, but the arduino.json does not support such option. I tried putting a 
```
"defines":[...]
```
in the c_cpp_properties.json, but it is not picked up during compilation.
Same goes for extra compile flags. I tried putting
```
"compilerArgs": [...]
```
in the c_cpp_properties.json, but it is not picked up.

elektronikworkshop, 2020-02-29:
Hi Slav again,

`c_cpp_properties.json` is a configuration file for IntelliSense and it's not related to the Arduino build back-end in any way. It's just for getting code highlighting and navigation right (This is what I'm implementing currently because how the extension manages it currently is pretty bollocks).

To cite [this post](https://forum.arduino.cc/index.php?topic=602603.0)
> There isn't really any convenient way to do this via the Arduino IDE. This was actually an intentional decision by Arduino because they feel it will lead library developers to unnecessarily create library interfaces that are not beginner friendly.

There is a way to pass flags to the compiler using the `compiler.extra_flags` preference. But you'd have to build your project from the command line:
```bash
arduino --pref compiler.extra_flags=-DF_USB=16000000 --upload --board arduino:avr:leonardo foobar.ino
```
And BTW: It's not a good idea to set this in Arduino's `preferences.txt` because if you do so, *all* sketches you build will have this flag set.

I suggest you try to work around it. What is the problem that forces you to pass a define to the compiler?

Regards,  
EW

elektronikworkshop, 2020-02-29:
Hi Slav,

I added support for additional build preferences in my branch. You can set any Arduino preferences to be used during build within `arduino.json`. Here's an example to set some build flags 
```json
{
    "board": "arduino:avr:nano",
    "sketch": "test.cpp",
    "port": "/dev/ttyUSB0",
    "configuration": "cpu=atmega328old",
    "buildPreferences": [
        ["build.extra_flags", "-DMY_DEFINE=666 -DANOTHER_DEFINE=3.14"],
        ["compiler.cpp.extra_flags", "-DYET_ANOTER=\"hello\""]
    ]
}
```
Within my branch those defines will then automagically be added to the IntelliSense configuration.

You can alpha test [here](https://www.dropbox.com/sh/whmcdt26chyjgby/AAB1Ld2fzZ9Z_NfM3CRay17wa).

Please note, that I'm not affiliated with the maintainers of this repository and my current work still has to be merged.

Regards,
EW

elektronikworkshop, 2020-02-29:
Thanks, @SlavDimov !

Your inquiry triggered me to implement this pretty neat feature!

Regards,
EW

SlavDimov, 2020-02-29:
Hi EW,
Awesome! Thank you for the quick patch!

To respond to your original question my use case is the following:
I am using .cpp and .h files instead of .ino. 
About the defines - I need to be able to exclude/include some code based on what I am doing. Say I have a "DEBUG" define and when it is enabled I want it to print a sensor reading every second on the serial monitor. When it isn't it will be excluded from the source.

About the compiler flags - I want to be able to pass extra arguments like -Wall, -Werror etc. Use cases are pretty straight forward here.

elektronikworkshop, 2020-02-29:
Hi @SlavDimov 

You're welcome!

vscode-arduino already supports `.cpp` main sketches. Just set the sketch to your `main.cpp` within `arduino.json`. All other `.cpp` sources within the sketch folder are compiled automatically as well. So no need to fiddle with compiler arguments in this case.

I usually write a library as soon as I have reusable code. This has several advantages:
* It remains portable
* no need for compiler arguments (again portability)
* others (or myself) can use it with the Arduino IDE

Actually I have some pretty complex projects which don't need any compiler fiddling. I can only recommend that for portability.

`DEBUG` defines: fair enough - I already suspected that it would be something like that. But that's nothing one couldn't solve with a global include.

The `-Wall` is globally enabled on my system within `~/.arduino15/preferences.txt` - good to see, that there are others who enable this. I can't stand folks which disable warnings :wink:

But anyhow: Having the power to set some preferences is a good thing and pretty powerful.

Regards,
EW

SlavDimov, 2020-02-29:
Hi EW,
I know about the CPP in arduino.json. I already did that. My point was that if you are using .ino files there is not much point in using defines. About the warnings, yes you can do that, but what happens when you collaborate with other people? It is better for them to live as an argument inside the .vscode folder

Regards,
Slav

elektronikworkshop, 2020-02-29:
Hi!
> I know about the CPP in arduino.json. I already did that. My point was that if you are using .ino files there is not much point in using defines.  

This might need more explanation (even though this might not be the right place to discuss such stuff).

Regarding the warnings: You mean because of that the bad code of others? I usually fix the warnings of others and file pull requests. But of course - having a project-wise switch at hand is pretty convenient. Especially for large frameworks which have a slow release cycle...

Happy coding!

Regards,
EW

ivankravets, 2020-03-02:
Please take a look at https://platformio.org that supports Arduino framework.

- https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide
- https://docs.platformio.org/en/latest/projectconf/section_env_build.html

elektronikworkshop, 2020-03-03:
Hi Ivan,

thanks, but there are people like me who want a lightweight extension and don't want to use Platform IO. Another of my concerns is its restricted licensing scheme.

Regards,  
EW

elektronikworkshop, 2020-03-03:
And BTW, @ivankravets : It's pretty uncool to come here to fish for people. This makes Platform IO for me even less attractive.

ivankravets, 2020-03-03:
> Another concern of me is its restricted licensing scheme.

@elektronikworkshop there is not difference will you use PlatformIO or won't. It's a fully free and open source. It's licensed under the maximum permissive Apache 2.0 license, where Arduino's new products including IDE are close sourced and even paid. Community Java-based Arduino IDE is under GPL.

I just wanted to save your time.

elektronikworkshop, 2020-03-03:
@ivankravets  Thanks for being so generous and saving my time!

Ah, I see - you changed your licensing... There seemed to be the simple mechanics at work: I don't like to contribute to projects where some guy then cashes in the revenues.

But what you're spreading here about Arduino is simply [wrong](https://blog.arduino.cc/2019/10/25/new-arduino-pro-ide-a-closer-look/). The new Arduino Pro IDE will be licensed under the EPL. But that doesn't matter, since vscode-arduino doesn't make use of it anyways. I think, the new Arduino Pro will render vscode-arduino and Platform IO obsolete for all the Arduino users. I wish you Good Luck!

ivankravets, 2020-03-03:
> Ah, I see - you changed your licensing... 

PlatformIO has been fully open source since the first day (2014) under Apache 2.0. We didn't change license.

> where some guy then cashes in the revenues.

What do you mean?

> But what you're spreading here about Arduino is simply

Are Arduino Create, Cloud and other newly added services free and open source? No, because there is no more freedom with Arduino. It has been acquired by ARM/Softbank.

> Platform IO obsolete for all the Arduino users. 

It is your personal opinion and I respect it. You can't create a product where 100% of people will be satisfied. There are hundreds of thousands of developers who are super happy with PlatformIO:
- https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide&ssr=false#review-details

> I wish you Good Luck!

Thanks! If you will need any help, do not hesitate to contact us.

elektronikworkshop, 2020-03-03:
Yes, the PIO's back-end was, but the IDE and other stuff [wasn't](https://community.platformio.org/t/what-part-of-platformio-is-open-source-licenced/1447/2).

Yeah, there are some services which are closed but the core will remain open. For advanced Arduino users Platform IO was an option due to the lack of professional features on the classic Arduino IDE. If this deficiency is gone, the demand for alternatives will perish. And I have the impression, that a considerable share of PIO's user base are simply Arduino users who had the same problem as I did.

ivankravets, 2020-03-03:
> Yes, the PIO's back-end was, but the IDE and other stuff wasn't.

Thanks, I've just edited that post! 

> If this deficiency is gone, the demand for alternatives will perish.

PlatformIO is not an alternative to Arduino. Arduino for us is a big Community of talented people and we work hard to keep full compatibility with legacy Arduino IDE builder. When I talk about Arduino, I don't mean "arduino.cc". I mean all the good things (Wiring) which were founded by Hernando. You can read more here [The Untold History of Arduino ](https://arduinohistory.github.io/). I prefer truth and justice. 

Today, PlatformIO is the only neutral, independent, free and open-source ecosystem for embedded development. We support over 30+ different architectures and dev-platforms. We don't force people to use the only Arm. Our mission to support as much as more different hardware and software solutions in the embedded market.

In the case of Arduino, this is a framework for us and is a small puzzle in our ecosystem. We support ~20 different frameworks and SDKs. Our customers have full freedom and personally decide which IDE or OS to use: VScode, Eclipse, CLion, etc ([see more](https://docs.platformio.org/en/latest/ide.html)). If we have tomorrow a much better text editor than VSCode, we will create a new extension and all previous projects will be compatible.

Just finalize. We don't compete or fight with Arduino. There is nothing common between Arduino and PlatformIO. They are totally different. We even have different business models. As I mentioned before, we respect the Arduino Community and help them as much as we can. PlatformIO - this is not about VSCode or debugging. This is about having a unified ecosystem for different architectures and dev-platforms. Our mission is to help silicon vendors to save their expensive resources working on "yet another Eclipse IDE". You can learn more about what we do from my recent talk at RISC-V Summit https://www.youtube.com/watch?v=0eYDKION0Bs
We help different silicon vendors which make products based on the different architectures.

P.S: I known, most of the developers look on PlatformIO as an enhancement for Arduino IDE. This is our personal mistake that we spent a lot of time on Arduino. Most of our advanced instruments, such as Project Inspection, Memory Analyzer, Static Code Analyzers, Unified Debugger are not intended for Makers. This is why I said that it is incorrect to compare Arduino vs PlatformIO. If your project is very simple with 1 file and you personally work on it - Arduino IDE is a great option for this. If you need advanced control on a build process or you have a team, they work on different machines, use a different OS, different IDEs - PlatformIO is ideal here with its support for hybrid configuration.

This issue is about "advanced control". This why I added 5 cents to save your time. Take a look at this project example https://github.com/MarlinFirmware/Marlin/blob/2.0.x/platformio.ini to understand what PlatformIO can.

MohammedNoureldin, 2020-10-30:
@elektronikworkshop , thank you for the nice implementation, will you send a pull request to merge this nice feature to the official extension in VS-Code?