microsoft/vscode-arduino #976: Can't compile functions with reference parameters

richard-mckenna, 2020-03-03:
I'm using the M5ez library and trying to use advanced functions with ezMenu addItem [https://github.com/ropg/M5ez#menus-all-the-functions-documented](https://github.com/ropg/M5ez#menus-all-the-functions-documented)

```c++
void mainmenu_manage_boards() {
    prefs.begin("BOARDSTATUS");
    String idx;
    ezMenu savedBoards("Managing Boards");
    if (board_count < 1) {
        ez.msgBox("No Boards", "You have no saved Boards.", "OK");
        return;
    }
    for (uint8_t n = 1; n < board_count+1; n++) {
        idx = "NAME" + (String)n;
        String name = prefs.getString(idx.c_str(), "");
        savedBoards.addItem(name, NULL, _savedSelected);
    }
    savedBoards.txtSmall();
    savedBoards.buttons("up#Back#Forget##down#");
    savedBoards.run();
    prefs.end();
}

bool _savedSelected(ezMenu* callingMenu) {
  if (callingMenu->pickButton() == "Forget") {
    if (ez.msgBox("Forgetting board", "Are you sure you want | to forget board | " + callingMenu->pickName() + " ?", "Yes##No") == "Yes") {
      callingMenu->deleteItem(callingMenu->pick());
    }
  }
  return false;
}
```

However compiling with the VSCode Extension returns the following error:

```
error: 'ezMenu' was not declared in this scope
  bool _savedSelected(ezMenu* callingMenu) {
```

Compiling with the Arduino IDE works fine.

elektronikworkshop, 2020-03-03:
Looks like you have some configuration not set up properly. The error message is not an error message of vscode-arduino but of Arduino's build backend. vscode-arduino uses Arduino's build backend, so there must be some differences in your setup. Most probably regarding your libraries. Your error log doesn't help that much. Better you verify with verbose enabled and take a look at the messages there.

richard-mckenna, 2020-03-04:
@elektronikworkshop interesting. I enabled verbose logging, exactly the same error. However I have figured out the fix.

So my main application .ino file was called **main.ino** changing this to **arduino.ino** I no longer see the error. Did not realise the .ino file needed to be named the same as it's parent folder.

Very misleading error message.

closed 2020-03-04 by richard-mckenna