microsoft/vscode-arduino #978: Project path cannot be changed

DrXoo, 2020-03-04:
I am writing a Sketch that uses a custom library that I wrote in the same folder.

At the moment of try verify the Sketch the compiles cannot find the header file (.h) for the library that I wrote.

In Arduino IDE this can be solved by changing the _project path._

Here we don't have anyway to change the project path for the compiler.

Here is my project structure right now
[image]

Inside the c_cpp_properties.json I have wrote the project path on both _Arduino_ and _Arduino-avr-standart_ configurations but got no results.

hellyzh, 2020-03-10:
Please check doc [Writing a Library for Arduino](https://www.arduino.cc/en/Hacking/libraryTutorial). And there is also a fix #946 (Support custom libraries with portable version of Arduino) to fix intellisense issue with custom libraries.

DrXoo, 2020-03-10:
> Please check doc [Writing a Library for Arduino](https://www.arduino.cc/en/Hacking/libraryTutorial). And there is also a fix #946 (Support custom libraries with portable version of Arduino) to fix intellisense issue with custom libraries.

Thanks @hellyzh I ended up dividing my project structure. Created a Repo for the library and used it on Documents/Arduino/Libraries and created another Repo for the .ino file

The main concern I was talking about was that the path for Libraries cannot be changed but this workaround is not so critical.

Thank you again

closed 2020-03-10 by DrXoo