microsoft/vscode-arduino #979: Cannot find Arduino IDE.

Ilie24118, 2020-03-05:
[image]

DrXoo, 2020-03-05:
Please execute the command Arduino: Initialize to create the arduino.json file. Inside you can specify the path

dilin-MS, 2020-03-20:
Hi @Ilie24118 , is the issue resolved for you?

Ilie24118, 2020-03-21:
no

hellyzh, 2020-03-26:
Arduino extension v0.3.0 is released to fix serial port issue which requires vscode 1.43, please try to open an arduino example using the extension.

El-Programador4, 2020-04-01:
I have the same problem as the author, and I don't know how to solve it!

hellyzh, 2020-04-02:
@El-Programador4, have you installed [Arduino IDE](https://www.arduino.cc/en/Main/Software) ?

Ilie24118, 2020-04-12:
Yes

hellyzh, 2020-04-13:
@Ilie24118 , Please try to set `arduino.path` to `C:\\Program Files (x86)\\Arduino` (the path of your Arduino IDE) and `arduino.commandPath` to `arduino_debug.exe` and restart your vscode.

closed 2020-05-21 by hellyzh