microsoft/vscode-arduino #980: Unable to use serial port in VSCode (Insiders) Version: 1.43.0

hellyzh, 2020-03-06:
Unable to use serial port in VSCode Insiders Version: 1.43.0.
Extension Version: 0.2.29
VSCode (Insiders) Version: 1.43.0 (system)
OS: Windows 10 (Clean installation a couple days ago)

VSCode Insiders 1.43.0 upgraded electron version, arduino extension also need to upgrade to support it.

phoenixxd, 2020-03-10:
I'm facing the same issue.

ck88988, 2020-03-10:
Same issue I found when I upgraded VSCode to V1.43. I returned to V1.42 till this issue can be solved.

Xring-git, 2020-03-10:
Same issue after upgrade to VSC 1.43.0

pilou64, 2020-03-10:
Same issue after upgrade to VSC 1.43.0
Downgrade to 1.42 is OK.
(don't forget to disable auto update on the settings)
https://stackoverflow.com/a/49347158

blfuentes, 2020-03-11:
Same hier:
VS Code 1.43.0 (system) 
OS: Win 10 1909 and Ubuntu 19.10

Downgrade to 1.42.1 makes it work again.

DoganM95, 2020-03-11:
Unfortunately a downgrade is never a persistent/good solution, as upgrades exist to close security vulnerabilities, fix bugs, etc.

The extension needs to be rewritten/fixed to work independently of vscode versions

DerekFoster1976, 2020-03-11:
Same here, cannot use 1.43 for MXChip development as serial port COM selector not longer functioning. Only fix is to downgrade to 1.42. I am assuming no testing has been done whatsoever with 1.43.

bzeeman, 2020-03-12:
Same here. What is the easiest process to downgrade?

blfuentes, 2020-03-12:
Go to https://code.visualstudio.com/updates/v1_42 and download the corresponding one

ophayden, 2020-03-13:
Same problem. When downgrading, must I uninstall my existing version of vscode before installing the old version?

ophayden, 2020-03-13:
Same issue after upgrade to 1.43

MikeX7, 2020-03-15:
How to make it work on 1.43:
Open arduino.json and add this line:
`"port": "COM3"`

blfuentes, 2020-03-15:
> How to make it work on 1.43:
> Open arduino.json and add this line:
> `"port": "COM3"`
> Obviously replace COM3 with the correct com port, you can find out which one it is by looking in the Arduino IDE

if you do so, you will be able to upload the sketch but the serial monitor doesn't work, or at least, it didn't work for me

yuhongshuai, 2020-03-15:
Same issue I found when I upgraded VSCode to V1.43. I returned to V1.42 till this issue can be solved.
yes

dilin-MS, 2020-03-19:
The root cause of the issue is because VS Code 1.43.0 use a new version of Electron, which is not compatible with our serialport. We have been working on fixing it and a new bit of arduino will soon come out.

rockboynton, 2020-03-19:
> The root cause of the issue is because VS Code 1.43.0 use a new version of Electron, which is not compatible with our serialport. We have been working on fixing it and a new bit of arduino will soon come out.

Thank you! I'm sure we all look forward to the update

hellyzh, 2020-03-26:
Arduino extension v0.3.0 is already released to fix serial port issue which requires vscode 1.43.

Awbmilne, 2020-03-26:
The update seems to have fixed it issue. Thanks @hellyzh and the rest of the Arduino extension team!

closed 2020-03-27 by hellyzh