microsoft/vscode-arduino #984: The Select serial port doenst work

JorgeDaniel369, 2020-03-13:
Issue Type: <b>Bug</b>

the serial dort dont work, im tryin to changue or select my port but it doesnt show me nothing

Extension version: 0.2.29
VS Code version: Code 1.43.0 (78a4c91400152c0f27ba4d363eb56d2835f9903a, 2020-03-09T19:47:57.235Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz (8 x 2808)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: enabled<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.89GB (10.08GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

blfuentes, 2020-03-13:
Take a look here:
https://github.com/microsoft/vscode-arduino/issues/980

dilin-MS, 2020-03-20:
Duplicate with #980

closed 2020-03-20 by dilin-MS