microsoft/vscode-arduino #988: Cannot select serial port

SuperNose, 2020-03-16:
Issue Type: <b>Bug</b>

Open the very lates Visual Code with very latest Arduino Extension and try to select the COM-Port - there is this error message:
Cannot read property 'settings' of null

Please, fix it, it's somehow important to do anything with the Arduino ;-)!

Thank you

Extension version: 0.2.29
VS Code version: Code 1.43.0 (78a4c91400152c0f27ba4d363eb56d2835f9903a, 2020-03-09T19:47:57.235Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz (12 x 2592)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: enabled<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.74GB (3.68GB free)|
|Process Argv||
|Screen Reader|no|
|VM|36%|
</details>

Rezaasghariir, 2020-03-17:
that's true i have that problem too.
Extension version: 0.2.29

MichaelCduBois, 2020-03-18:
Confirmed issue:

VS Code Version: 1.43.0
Electron: 7.1.11
Extension Version: 0.2.29

dilin-MS, 2020-03-20:
Dulplicate with #980

closed 2020-03-20 by dilin-MS