microsoft/vscode-arduino #989: Similar to issue #927 - no serial port choice for VS Code

micahalan, 2020-03-18:
I am currently having this same issue as issue #927  with version 1.43.0

The serial port "link" does nothing when I click on it, or try and reach that functionality through other menus.

rneurink, 2020-03-18:
Same issue here. I can select boards but the serial port selection trough the status bar or through the command does not work a.t.m.

Jojojdotpng, 2020-03-18:
You can try downgrading VSCode to version [v1.42](https://code.visualstudio.com/updates/v1_42). it  worked fine on my machine

micahalan, 2020-03-19:
Thanks @Jojojdotpng !! Good idea and it seems to have worked.

BinaryHole, 2020-03-19:
I am having the same issue.
VSCode - 1.43.1
Electron - 7.1.11

rockboynton, 2020-03-19:
Same issue 1.43.1

dilin-MS, 2020-03-20:
Dulpicate with #980

closed 2020-03-20 by dilin-MS