microsoft/vscode-arduino #994: Arduino version 1.8.12

davewilk, 2020-03-20:
The extension installation instructions say

Note: Arduino IDE 1.8.7 has some breaking changes, causing board package and library installation failures. It is recommended to that you install version 1.8.6 or 1.8.8

But current Arduino IDE version is 1.8.12. Am I not supposed to use that?

hellyzh, 2020-03-23:
1.8.12 is OK. It supports 1.8.8 or later.

davewilk, 2020-03-23:
OK, thanks, but why is that directive still present in the installation instructions? I  have to say that I am rather disappointed in this extension, because I also ran into the "Unable to set COM port" bug. Even after setting the port in the .json file the serial; port is not not available for use, which makes the extension a non-starter for many/most projects.

Otherwise it would be great -- after using Visual Studio and Visual Studio Code for many years, the Arduino IDE is a real pain to use.

hellyzh, 2020-03-26:
Arduino extension v0.3.0 is released to fix serial port issue which requires vscode 1.43.

davewilk, 2020-03-26:
OK, good. Thanks!

hellyzh, 2020-03-27:
Thank you for your suggestion. It already mentioned in the prerequistes that the supported Arduino IDE versions are 1.6.x and later.

davewilk, 2020-03-27:
Yes, but the first bullet in the instructions still says:

Note: Arduino IDE 1.8.7 has some breaking changes, causing board package and library installation failures. It is recommended to that you install version 1.8.6 or 1.8.8

which to me was very confusing. If the extension supports 1.6 and later except 1,87 then this is what it should say.

hellyzh, 2020-03-28:
Yes, the first notice is out of date especially the recommendation of installing version. We will update this.

hellyzh, 2020-04-02:
#1012 submitted the update

closed 2020-05-21 by hellyzh