microsoft/vscode-arduino #998: Serial Monitor stop working after vscode v1.43 update

flautines, 2020-03-22:
Issue Type: <b>Bug</b>

Steps to reproduce the issue:
At the Arduino extension bottom bar click on the "Open Serial Monitor" button.

expected result:
Serial communication is started with the selected compatible board and at the specified COM port.

actual result:
The following error is shown:
Failed to open serial port COM4 due to error: + TypeError: Cannot read property 'close' of null

I made sure the selected port and board was correct by opening the Serial Monitor at the selected port and board from the Arduino IDE.

Extension version: 0.2.29
VS Code version: Code 1.43.1 (fe22a9645b44368865c0ba92e2fb881ff1afce94, 2020-03-18T07:01:20.184Z)
OS version: Windows_NT x64 10.0.18362

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz (8 x 4008)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.96GB (11.69GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

ZiTAL, 2020-03-22:
https://github.com/Microsoft/vscode-arduino/issues/774

dilin-MS, 2020-03-23:
Duplicate as #980

closed 2020-03-23 by dilin-MS