microsoft/vscode-arduino #1008: Cannot use Serial Monitor

c0range, 2020-03-26:
VSCode - Latest (non insider)
Arduino - 0.2.29

When using the Arduino extension with an Adafruit ESP32 Feather (COM3, 80hz, upload speed 921600, serial baud 115200), attempting to open Serial Monitor in VSCode returns the following error.

`Failed to open serial port COM3 due to error: + TypeError: Cannot read property 'close' of null`

This happens every time and I cannot get Serial Monitor to work within VSCode. Any idea as to why?

(also, I am new here and this is my first issue. Please forgive me if I forgot to include something.)

hellyzh, 2020-03-26:
Arduino extension v0.3.0 is released to fix serial port issue which requires vscode 1.43.

hellyzh, 2020-03-27:
Duplicate of #980

closed 2020-03-27 by hellyzh