microsoft/vscode-arduino #1011: Visual Code can´t find the Arduino IDE

El-Programador4, 2020-04-01:
I don't know how to solve it, visual code  can´t find the Arduino IDE.
[image]

hellyzh, 2020-04-02:
Please set `arduino.commandPath` to `arduino_debug.exe` and `arduino.path` to `G:\Arduino`, and restart your vscode.
[image]
[image]

hellyzh, 2020-04-02:
#979 Visual Code can´t find the Arduino IDE

El-Programador4, 2020-04-02:
That doesn't work for me
[image]

hellyzh, 2020-04-03:
change it to `"arduino.path": "G\\Arduino\\arduino-1.8.12"`

Iyaka25747, 2020-04-10:
I have the same issue but work with Unbuntu. Could you help me to figure out what arduino.path shall I use to bypass this issue ?

hellyzh, 2020-04-14:
@Iyaka25747 , thank you for confirm working. If no more issue, will close this issue.

rohit-singh-gautam, 2020-05-02:
I installed Arduino using Windows Store, now unable to find the path of Arduino, how to resolve this?

hellyzh, 2020-05-21:
Arduino using windows store is not supported, you need to install Arduino IDE.

closed 2020-06-23 by hellyzh