microsoft/vscode-arduino #1013: Extension causes high cpu load

k1215tian, 2020-04-02:
- Issue Type: `Performance`
- Extension Name: `vscode-arduino`
- Extension Version: `0.3.0`
- OS Version: `Windows_NT x64 10.0.10586`
- VSCode version: `1.43.2`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`C:\Users\mis-dev\vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

rrakso, 2020-05-19:
Here the same
[vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt](https://github.com/microsoft/vscode-arduino/files/4651168/vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt)

hellyzh, 2020-06-09:
Arduino extension v0.3.1 is released.

closed 2020-06-09 by hellyzh