microsoft/vscode-arduino #1015: Serial Monitor keeps ESP devices in flash mode

vlastahajek, 2020-04-07:
__Visual Studio Code__: 1.43.2
__Arduino Extension__: 0.3.0

__Isssue__: Opened Serial Monitor keeps ESP8266 and ESP32 in flash mode.

__Details__:
ESP8266 after pressing reset twice:
```
ets Jan  8 2013,rst cause:2, boot mode:(1,6)

ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

ESP32 after pressing reset twice:
```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

Close Serial Monitor and re-plug USB cable and device works OK.

Downgrading to VSC 14.2.1 and AE 0.2.29 fixes this.

hellyzh, 2020-04-08:
There is no such issue for other device like Devkit AZ3166. And no issue on Linux or Mac OS for ESP32. It looks like related to unreliable reset on windows for ESP32 development boards, please check [doc](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection) to see if it helps.

vlastahajek, 2020-04-08:
I understood that this issue is weird. I have pre-manufactured boards (I apologize for forgetting to add board info, now it is fixed) and I didn't have this problem before upgrading to AE 0.3.0.
I did the cycle with downgrading to VSC 1.42.1 +AE 0.2.29 to resolve that and upgrading again to VSC 1.43.2 + AE 0.3.0 to verify the issue persist twice.

I will try this another computer to verify if it is a problem with the current computer.

mathertel, 2020-04-19:
Some analysis:

It is about the additional Serial signals beside RX and TX. The current Arduino Extension (version 0.3.0) behaves different on the DTS and RTS signals that results in pulling the GPIO0 to low. Under this conditions a reset will start flash mode. This is reported by boot mode:(1,6) (use 74880 baud)
See:
https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes for more details.

Using the Arduino IDE and pressing reset while the Serial Monito is open reports: boot mode:(3,6)

Conclusion: The current Arduino Extension behaves different that the former versions and the Arduino IDE regarding the Serial interfaces (DTS and RTS).

alarobric, 2020-05-01:
Noticed the same issue today. When using the Arduino IDE I see no issue, but using VSCode and this extension, if I reset my board while the serial monitor is open I just get stuck in boot mode.

This similar issue points to DTR being handled differently in these different serial monitor tools and gives the same result mathertel found above: 
https://github.com/serialport/node-serialport/issues/1854#issuecomment-496887057

Is there any way to adjust this behaviour in the serial monitor?

larsschellhas, 2020-05-01:
As the previous commentators, I've just encountered the same issue with my Wemos D1 R2.
@hellyzh has this been fixed in the dev branch as you've closed the issue?

thebigpotatoe, 2020-05-13:
Same issues here updating to the latest everything using a NodeMCU as my main board. The roll back solution works, but I have other projects which now require the latest IDE, so it isn't possible for me to use it anymore.

A dirty hardware solution i found was to connect a NC push button to GPIO 2 which you would then click when you wanted to upload to the board. This holds the ESP in the correct boot mode other times except upload.

mathertel, 2020-05-13:
Some ugly workarounds:
* use the Arduino IDE to upload and monitor but not for editing. There is an "external editor" option you should switch on.
* never manually reset while the monitor is open but disconnect, press reset and re-connect. I added a delay(3000) at the start of my sketch to see everything.
* wait for someone to fix the -b-u-g- problem.

dilin-MS, 2020-05-21:
Hi, the latest serialport lib we used causes the issue, see [serialport issue](https://github.com/serialport/node-serialport/issues/1854) and [hupcl](https://github.com/serialport/node-serialport/issues/1678). And we have discovered a [workaround](https://github.com/serialport/node-serialport/issues/1854#issuecomment-496887057) to fix it.
Currently we are testing the solution. If it works, a new bit will be released to solve the problem. Thank you very much.

hellyzh, 2020-06-09:
Arduino extension v0.3.1 is released to fix the issue.

closed 2020-06-09 by hellyzh

iqapps, 2020-06-10:
After this update, I cannot DTR/RTS reset Arduino Mega Embed from Robodyn (CH340) by serial port disconnect/connect anymore. It has othervise worked perfectly up to this update.