microsoft/vscode-arduino #1016: Plugging an an arduino with VSCode open to a non-arduino project causes c++ and arduino files to be added to that project

wilsonmfg, 2020-04-08:
the files are:
.vscode/arduino.json
.vscode/c_cpp_properties.json

these really do not belong in a nodejs or python project.  the extension should check for an .ino file before adding them.

hellyzh, 2020-04-10:
Is this because you used the command `"Arduino: Examples"` to open the project? If use `Open Folder...` to open the project, it should not add the files.

wilsonmfg, 2020-04-10:
no, the project in question is not an arduino project at all.

adamm, 2020-04-10:
I have the same issue.

It occurs when I plugin an [esp32 dev kit](https://www.espressif.com/en/products/hardware/esp32-devkitc/overview) and /dev/cu.SLAB_USBtoUART is created.   It does not occur when I plug in an Arduino Uno or Nano (which create /dev/cu.usbserial-xxxxxxxx).

[image]

What's interesting, when this issue occurs and my non-Arduino project is "hijacked" by the VS Arduino extension, the esp32 is being mis-identified as an esp8266, and the [esp8266 README.md](https://github.com/esp8266/Arduino/blob/master/README.md) opens in a new tab.

@wilsonmfg what hardware are you connecting?

Mac OS: 10.13.6
Visual Studio: 1.44.0
VSCode Arduino: 0.3.0

wilsonmfg, 2020-04-13:
an adafruit Huzzah ESP8266.  
is it possible there some code in the esp8266 board definition that's triggering vscode to do this?

hellyzh, 2020-04-14:
Mis-identified esp32 is due to a hard code [usb mapping](https://github.com/microsoft/vscode-arduino/blob/master/misc/usbmapping.json#L110), It may need to manually switching board after plugged device in. But for the wrongly added arduino files, they really should be only added to project with ".ino" file.

adamm, 2020-04-15:
I've found that I can workaround the issue and avoid the hijack by adding to global settings.json:
`    "arduino.enableUSBDetection": false`
And restarting vscode.

hellyzh, 2020-06-09:
Arduino Extension v0.3.1 is released with the fix to not add arduino files for non-Arduino project.

closed 2020-06-09 by hellyzh