microsoft/vscode-arduino #1019: WiFiClientSecure has no member "verify"

hasanemretonguc, 2020-04-13:
am i doing something wrong ?

[image]

hellyzh, 2020-04-13:
It looks like the board package issue. Have you selected the right board?
[image]

hasanemretonguc, 2020-04-13:
im using nodemcu v3 board. i change to generic esp8266. issue still persist
[image]

hellyzh, 2020-04-14:
You can go to definition of WiFiClientSecure to check if there is member "verify". It should be other member like "connected", "available".

hasanemretonguc, 2020-04-14:
looks like it is
[image]

hellyzh, 2020-04-15:
Is `"Arduino: Verify"` or `"Arduino: Upload"` working or not? If yes, it may be intellisense issue.

hasanemretonguc, 2020-04-15:
verify and upload works

hellyzh, 2020-04-16:
It should be issue for the intellisense, please check the setting of file .vscode/c_cpp_properties.json according to [link](https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference).

hasanemretonguc, 2020-04-17:
i couldnt find the problem can you help with that 
[image]

hellyzh, 2020-04-21:
Please also try to specify `"compilerPath"`

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay