microsoft/vscode-arduino #1021: Cannot connect to Serial port.

navanshu, 2020-04-19:
### Cannot open serial port monitor, also cannot upload the .ino files to the board.

I am using ubuntu WSL latest and arduino plugin in vs code. I am new to using this plugin in WSL. I have installed arduino IDE and have setup vscode to work with it.
Here is my arduino.json
Arduino is connected to COM port 6.
```javascript
{
    "board": "arduino:avr:uno",
    "port": "/dev/ttyS6/",
    "sketch": "app/ardtest.ino",
    "output": "../build",
    "debugger": "jlink",
    "programmer": "AVRISP mkII"
}
```
and here is my settings.json
```javascript
{
    "arduino.path": "/home/rohan/arduino-1.8.12",
    "arduino.commandPath": "arduino",
    "arduino.defaultBaudRate": 9600
}
```

When I am trying to open the serial port, I am getting this error:
Failed to open serial port /dev/ttyS6/ due to error: + Error: The module '/home/rohan/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.0/out/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 67. This version of Node.js requires NODE_MODULE_VERSION 72. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).

I also followed to this documentation on microsoft website on how to connect to serial port through WSL -> [Serial Support on the Windows Subsystem for Linux](https://docs.microsoft.com/en-us/archive/blogs/wsl/serial-support-on-the-windows-subsystem-for-linux)

Please tell me the steps I should take to make this work. I am not sure what is the issue I am facing here, will compiling the plugin will solve my problem. If there is an alternative way I can communicate with the boad.

hellyzh, 2020-04-26:
You can try to upgrade version of [nodejs](https://nodejs.org/en/download/).

korompaiistvan, 2020-05-26:
I just ran into the same exact issue. I'm running Ubuntu on WSL and getting the same errors. node was on latest version (v14.3.0) throughout the installation.

MitchTalmadge, 2020-06-27:
Looks like this is the root cause: https://github.com/MadLittleMods/node-usb-detection/issues/87

Not sure which package used by this repo depends on `usb-detection` yet.

MitchTalmadge, 2020-06-27:
`usb-detection` is required by `node-usb-native`. The latest version of `usb-detection` is `4.7.0` but the latest is `4.9.0`. This repo's package lock is set to `4.7.0`:

https://github.com/microsoft/vscode-arduino/blob/947744ddb5dabaacf3cb40b8091f4194ee2637af/package-lock.json#L8603-L8612

MitchTalmadge, 2020-06-27:
I temporarily solved this by running this script one line at a time. Basically I update the package-lock and then swap the new one out with the old, and do a new install. This updates the dependencies.

```
npm i --package-lock-only --prefix ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.1

mv ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.1/out/package-lock.json ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.1/out/package-lock.json.bak

mv ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.1/package-lock.json ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.1/out/package-lock.json

npm i --prefix ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.3.1/out
```

robotdad, 2020-10-30:
I just pushed an update to the extension, 0.3.3, that revved the usb-detection dependency. I'm not confident that is the underlying issue here based on the description, but could you try the new release and let us know if it is still an issue?

anuragmangla, 2020-12-26:
The issue persists in extension v0.3.4 with VSCode v1.52.1.

flyandi, 2021-04-01:
Still broken in v0.4.0 
[image]

unsplorer, 2021-05-08:
I was having the same issue in WSL VS code 1.56.0 with Arduino 0.4.2. Solved by using nvm ( https://github.com/nvm-sh/nvm ) to install node v14.16.0 then running 'npm i ~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/' and '~/.vscode-server/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/'. The select serial device is still hit and miss, so you'll need to add "port" : "/dev/WHATEVERYOURSERIALPORTISHERE" to the arduino.json file in your project rootdir .vscode folder.

oliveirajoaop, 2021-11-12:
Hi

I found a solution for a correct detection and usage of usb devices (arduino/usb-serial ch340 ) in wsl2/code and it can be found in this [link](https://kumekay.com/compiling-custom-kernel-for-wsl2/)

gcampbell-msft, 2023-02-16:
@navanshu We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue!

navanshu, 2023-02-20:
@gcampbell-msft I have been very busy and will be unable to tell you whether it works or not.
Thanks for the update, I will let you know, when I get the chance.

closed 2023-04-07 by github-actions[bot]