microsoft/vscode-arduino #1024: VSCODE - Arduino (MS) chooses wrong port under linux/chorme os

leemke, 2020-04-21:
Issue Type: <b>Bug</b>

Running latest VSCODE and arduino 1.8.8 in Chrome OS (under linux buster).

My initial setup code "COM1" as the default port value. And it would not allow me to change it.  COM1 does not mean a thing in linux.  I ran the arduino IDE and every thing works correctly.  

I found out that if I add '  "port": "/dev/ttyAMC0" to the .arduino.json file it will use that value after restarting and everyting works under vscode/arduino.

Please fix the the selecting the correct OS or allow the users to set the port via an easy way without having to spend hours figuring out how to do thi

Extension version: 0.3.0
VS Code version: Code 1.44.2 (ff915844119ce9485abfe8aa9076ec76b5300ddd, 2020-04-16T17:50:03.709Z)
OS version: Linux x64 4.19.98-08076-g24ab33fb8e14

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|06/8e (2 x 1607)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: disabled_off<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: disabled_software<br>skia_renderer: disabled_off_ok<br>video_decode: unavailable_off<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|0, 0, 0|
|Memory (System)|6.48GB (2.83GB free)|
|Process Argv|--no-sandbox --unity-launch|
|Screen Reader|no|
|VM|100%|
</details>

hellyzh, 2020-04-26:
You can run command `"Arduino: Select Serial Port"` or click the icon of Serial Port to select correct serial port.
[image]

leemke, 2020-05-01:
I have tried both. Neither works. The only way I found that works is to add
the "port":"/dev/ttyACM0", 
to the Adruino.json file

closed 2023-04-02 by github-actions[bot]