microsoft/vscode-arduino #1028: Serial monitor failed when restart the Arduino board

mynameisbill2, 2020-04-30:
Serial monitor failed when I restart the Arduino board(ESP8266), like this ...

[image]

But the same action in Arduino IDE is OK!

[image]

- Arduino IDE Version: 1.8.12
- VS Code Version: 1.43.2
- VS Code Arduino Extension Version: 0.3.0
- OS: Win8.1 Pro

hellyzh, 2020-04-30:
It looks like the same issue as #[1015](https://github.com/microsoft/vscode-arduino/issues/1015)(Serial Monitor keeps ESP devices in flash mode), Will serial monitor work if unplug and re-plug the ESP8266?

mynameisbill2, 2020-04-30:
Yes, It's the same problem.

closed 2020-04-30 by mynameisbill2