microsoft/vscode-arduino #1031: Error - cannot open source file "avr/pgmspace.h"

cvmanjoo, 2020-05-08:
Hi Someone can help?
I'm getting this error. Code complies fine but intellisense does not.

```
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\Manjunath CV\Documents\Arduino\sketch_may08b\sketch_may08b.ino).C/C++(1696)
cannot open source file "avr/pgmspace.h" (dependency of "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Arduino.h")C/C++(1696)
```

Thanks

hellyzh, 2020-05-21:
Please check your IntelliSense setting according to https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference.

ar-std, 2020-05-24:
I do have the same problem and already posted a more explanatory question on [stack**overflow**](https://stackoverflow.com/questions/61954481/intellisense-throws-include-errors-when-working-with-arduino-in-vs-code). Maybe this helps.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay