microsoft/vscode-arduino #1033: Problem in includePath with custom libraries, Verify/Upload doesn't work.

brunotacca, 2020-05-17:
Hello,

I decided to open an issue after some days trying it out, unfortunately, I am still stuck with this.

The basic examples available by Arduino IDE works fine, Blink for example, does upload/verify without a problem.

But, with custom libraries, it doesn't work. Although it works uploading it via Arduino IDE.
For my case, it's STBLE library, and UARTPassThrough example.

At start, it was showing intellisense errors about includePath, etc. After some tweaks  the browsePath worked fine (ctrl+click goes to the files), no errors shown by intelliSense, but when I click VERIFY, like if the classes/methods/definitions from the libraries are not loaded.

I get errors like:

> UART:24:3: error: 'HCI_Init' was not declared in this scope
>    HCI_Init();
> UART:26:3: error: 'BNRG_SPI_Init' was not declared in this scope
>   BNRG_SPI_Init();

Even tho, they are found just fine by the browse.path (ctrl+click)

The STBLE library is at the user documents, like:
> C:\Users\Tacca\Documents\Arduino\libraries\STBLE\src\arduino_bluenrg_ble.cpp

Even adding the custom library to includePath doesn't work.
```
"C:\\Users\\Tacca\\Documents\\Arduino\\libraries\\STBLE\\**"
```

**Thank you all in advance.**

My specs:
- Windows 10
- MingW x64 (at c:/MinGW64, env path works fine at powershell) 
- Arduino IDE 1.8.12
- VSCode 1.45.1
- - s-vscode.cpptools 0.28.0.
- - vsciot-vscode.vscode-arduino 0.3.0

settings.json:
```
{
    "dart.debugExternalLibraries": false,
    "dart.debugSdkLibraries": false,
    "window.zoomLevel": 0,
    "dart.lineLength": 120,
    "dart.openDevTools": "flutter",
    "C_Cpp.intelliSenseEngine": "Tag Parser",
    "C_Cpp.default.intelliSenseMode": "gcc-x64",
    "C_Cpp.loggingLevel": "Debug",
    "arduino.additionalUrls": "http://files.tinycircuits.com/ArduinoBoards/package_tinycircuits_index.json",
    "arduino.path": "C:\\Program Files (x86)\\Arduino", 
    "arduino.logLevel": "info", 
    "arduino.enableUSBDetection": true, 
    "C_Cpp.commentContinuationPatterns": [
        "/**"
    ],
}
```

arduino.json:
```
{
    "sketch": "UART.ino",
    "port": "COM5",
    "board": "TinyCircuits:samd:tinyzero",
    "configuration": "BuildOption=normal",
    "programmer": "AVR ISP"
}
```

c_cpp_properties.json:
```
{
    "configurations": [
        {
            "name": "Win32",
            "defines": [
                "ARDUINO=10800"
            ],
            "includePath": [
                "${workspaceRoot}",
                "C:\\Users\\Tacca\\AppData\\Local\\Arduino15\\packages\\TinyCircuits\\hardware\\samd\\1.1.0\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "C:\\Users\\Tacca\\Documents\\Arduino\\hardware\\tools\\**",
                "C:\\Users\\Tacca\\Documents\\Arduino\\libraries\\**"
            ],
            "forcedInclude": [
                "C:\\Users\\Tacca\\AppData\\Local\\Arduino15\\packages\\TinyCircuits\\hardware\\samd\\1.1.0\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "gcc-x64",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "browse": {
                "path": [
                    "${workspaceRoot}",
                    "C:\\Users\\Tacca\\AppData\\Local\\Arduino15\\packages\\TinyCircuits\\hardware\\samd\\1.1.0\\**",
                    "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**",
                    "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                    "C:\\Users\\Tacca\\Documents\\Arduino\\hardware\\tools\\**",
                    "C:\\Users\\Tacca\\Documents\\Arduino\\libraries\\**"
                ],
                "limitSymbolsToIncludedHeaders": false,
                "databaseFilename": ""
            },
            "compilerPath": "C:\\MinGW64\\bin\\gcc.exe"
        }
    ],
    "version": 4
}
```

hellyzh, 2020-05-21:
Please refer to [Writing a Library for Arduino](https://www.arduino.cc/en/Hacking/libraryTutorial) to write custom library. "c_cpp_properties.json" will not impact the VERIFY result, it is only for language check.

brunotacca, 2020-05-21:
> Please refer to [Writing a Library for Arduino](https://www.arduino.cc/en/Hacking/libraryTutorial) to write custom library. "c_cpp_properties.json" will not impact the VERIFY result, it is only for language check.

It's not a problem with the library, but maybe a compiler OR path problem.
I can state that fore sure because it works fine on Arduino IDE. But, at VSCode VERIFY/UPLOAD, it throws those "declaration not found" errors.

I am not stuck, because atm I am using VSCode to code, but to upload I need to open Arduino IDE every time, so is kinda annoying.

hellyzh, 2020-05-26:
Please remove `"C_Cpp.intelliSenseEngine": "Tag Parser"` or set it as `"C_Cpp.intelliSenseEngine": "Default"`, then the intellisense will give error. Have you include 'src/arduino_bluenrg_ble.h'

brunotacca, 2020-05-26:
> Please remove `"C_Cpp.intelliSenseEngine": "Tag Parser"` or set it as `"C_Cpp.intelliSenseEngine": "Default"`, then the intellisense will give error. Have you include 'src/arduino_bluenrg_ble.h'

After doing that, I get the following errors (error message translated)... also, the src/arduino_bluenrg_ble.h is located at "C:\\Users\\Tacca\\Documents\\Arduino\\libraries\\STBLE\\src\\STBlueNRG\\**"
even if I include this folder, it doesn't work... 

After digging a little more, I found that these errors pops up after open a file and at the first code line, always.

> #include error detected. Update your includePath. Wavy lines are disabled for this translation unit...
and
> The file "Source" cannot be opened: "sam.h".

```
{
	"resource": "/c:/Users/Tacca/Desenvolvimento/git/knee_kinematic_sensor/SensorCoreApp/SensorCoreApp.ino",
	"owner": "C/C++",
	"code": "1696",
	"severity": 8,
	"message": "#incluir erros detectados. Atualize o includePath. Os rabiscos estão desabilitados para esta unidade de tradução (C:\\Users\\Tacca\\Desenvolvimento\\git\\knee_kinematic_sensor\\SensorCoreApp\\SensorCoreApp.ino).",
	"source": "C/C++",
	"startLineNumber": 16,
	"startColumn": 1,
	"endLineNumber": 16,
	"endColumn": 18
}

{
	"resource": "/c:/Users/Tacca/Desenvolvimento/git/knee_kinematic_sensor/SensorCoreApp/SensorCoreApp.ino",
	"owner": "C/C++",
	"code": "1696",
	"severity": 8,
	"message": "não é possível abrir o arquivo fonte \"sam.h\" (dependency of \"C:\\Users\\Tacca\\AppData\\Local\\Arduino15\\packages\\TinyCircuits\\hardware\\samd\\1.1.0\\cores\\arduino\\Arduino.h\")",
	"source": "C/C++",
	"startLineNumber": 16,
	"startColumn": 1,
	"endLineNumber": 16,
	"endColumn": 18
}
```

Then again, I went to find "sam.h", found it at 2 paths:
                "C:\\Users\\Tacca\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0\\CMSIS\\Device\\ATMEL\\samd51\\include\\**",
                "C:\\Users\\Tacca\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0\\CMSIS\\Device\\ATMEL\\**"

Tried to import then both and each one separately, new errors popped up.

```
{
	"resource": "/c:/Users/Tacca/AppData/Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/samd51/include/sam.h",
	"owner": "C/C++",
	"severity": 8,
	"message": "diretiva #error: Library does not support the specified device",
	"startLineNumber": 50,
	"startColumn": 2,
	"endLineNumber": 50,
	"endColumn": 7
}
```

and a lot of "not defined" errors, at my workspaceRoot.

```
o identificador "SerialUSB" não está definido
{
	"resource": "/c:/Users/Tacca/Desenvolvimento/git/knee_kinematic_sensor/SensorCoreApp/SensorCoreApp.ino",
	"owner": "C/C++",
	"severity": 8,
	"message": "o identificador \"BLEsetup\" não está definido",
	"startLineNumber": 67,
	"startColumn": 3,
	"endLineNumber": 67,
	"endColumn": 11
}
o identificador "aci_loop" não está definido
o identificador "SerialUSB" não está definido
o identificador "SerialUSB" não está definido
o identificador "lib_aci_send_data" não está definido
o identificador "lib_aci_send_data" não está definido
```

I still can't figure it out since it works fine on Arduino IDE. not just that, the "ctrl+click" works fine, it finds and goes to the references... I will try a cold reinstall when I get a stable version of my little project. 
Thanks.