microsoft/vscode-arduino #1038: Arduino Serial

Puneetkumarpal, 2020-05-28:
Issue Type: <b>Bug</b>

serial prints nothing after reset tha esp8266 module.

but it works on program update

Extension version: 0.3.0
VS Code version: Code 1.45.1 (5763d909d5f12fe19f215cbfdd29a91c0fa9208a, 2020-05-14T08:27:35.169Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz (4 x 2594)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.90GB (1.68GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

hellyzh, 2020-06-01:
#1015 Serial Monitor keeps ESP devices in flash mode

hellyzh, 2020-06-09:
Arduino extension v0.3.1 is released to fix the issue.

closed 2020-06-09 by hellyzh

Puneetkumarpal, 2020-06-24:
Thanks IT's Working now