microsoft/vscode-arduino #1040: Pressing reset button while connected to VS Code serial monitor will brick ESP32 until power removed or sketch reuploaded

andrewzli, 2020-05-31:
- VS Code version: 1.45.1
- OS version: Windows 10 Pro v1909
- Microcontroller: ESP32 NodeMCU-32s

I have an issue that occurs when I press the reset button while reading from the serial monitor within VS Code. After pressing the reset button while reading from the serial monitor, the esp32 sketch code will stop functioning, and nothing can be read from the serial monitor. The only solution is to unplug and replug the ESP32, or re-upload the sketch. This issue happens regardless of the sketch code and can be replicated using a simple blink sketch that prints to the serial monitor (I've attached a quick sketch below as well). 

Steps of replication:

1. Create sketch, initialize the arduino folder within VS Code
2. Set the board to NodeMCU-32s (the same issue will occur regardless of board type I've found)
3. Connect the esp32, select the serial port, and upload the sketch
4. Set the serial monitor baud rate to match the rate specified in the sketch
5. Open the serial monitor

At this stage, the light on the ESP32 is now blinking and every second "Hello" is printed to the serial monitor, as expected. 

6. Press the reset (EN) button on the ESP32

Now no matter what I do, the LED on the ESP32 stays off and nothing will print to the serial monitor.

Another interesting complication is that if I do steps 1 to 5 but before pressing the reset button, I close the serial monitor, and press the reset button, the same issue occurs: no blinking light.

I have used the **same sketch code within the Arduino IDE using an ESP32**, as well as using the same code on an Arduino UNO while using VS Code. No issues, works like a charm in both cases. **This problem only occurs when I use an ESP32 in VS Code.**

If anyone has any ideas of things I may be doing wrong please let me know!

**esp32-blink.ino**
```
int ledPin = 2;

void setup() {
  pinMode(ledPin, OUTPUT);
  Serial.begin(9600);
  Serial.println("Starting...");
}

void loop() {
  Serial.println("Hello");
  digitalWrite(ledPin, HIGH);
  delay(500);
  digitalWrite(ledPin, LOW);
  delay(500);
}
```
**c_cpp_properties.json**
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\andre\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\**",
                "C:\\Users\\andre\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4\\**"
            ],
            "forcedInclude": [],
            "intelliSenseMode": "msvc-x64",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```
**arduino_properties.json**
```
{
    "sketch": "esp32-blink.ino",
    "board": "esp32:esp32:nodemcu-32s",
    "configuration": "FlashFreq=80,UploadSpeed=921600",
    "port": "COM6"
}
```

hellyzh, 2020-06-01:
#1015 Serial Monitor keeps ESP devices in flash mode

hellyzh, 2020-06-09:
Arduino extension v0.3.1 is released to fix the issue.

closed 2020-06-09 by hellyzh