microsoft/vscode-arduino #1041: Support VS Code on Windows on ARM

dilin-MS, 2020-06-01:
Related to [98818](https://github.com/microsoft/vscode/issues/98818)

To support VS Code on Windows on ARM, native node in `node-usb-native`, `serialport`, `usb-detection` will need to be rebuilt.

robotdad, 2020-10-29:
VS Code is available on ARM now. I need to take a look at what would be required for the underlying dependencies here to see if we can enable this or not.

maddogjt, 2020-10-30:
@robotdad happy to help take a look at this if you'd like, can test on raspberry pi, which I assume is a likely target device for this feature.

robotdad, 2020-10-30:
Thanks! I'll let you know if (when) we decide to take this on.

robotdad, 2020-10-30:
Adding the serial tag here as the native components that need to be compiled for arm are for serial port access

closed 2023-04-02 by github-actions[bot]