microsoft/vscode-arduino #1051: Extension causes high cpu load

rrakso, 2020-06-09:
- Issue Type: `Performance`
- Extension Name: `vscode-arduino`
- Extension Version: `0.3.1`
- OS Version: `Linux x64 5.4.0-33-generic`
- VSCode version: `1.45.1`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`/home/oskar/vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

rrakso, 2020-06-09:
[vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt](https://github.com/microsoft/vscode-arduino/files/4753702/vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt)

robotdad, 2020-10-30:
Can you provide some more info on when the extension is causing high cpu load? Is it all the time, just during builds, etc.?

BensnGit, 2020-12-03:
Hello, 
i have the same issue.
The cpuprofile file is attached but the issue is still here.

The problem always occurs immediately after uploading.
Im working on a M5Stack Esp32.

Aravindray, 2021-05-16:
Extension causes high cpu load
Same issue, but I am using Windows 10
What to do?