microsoft/vscode-arduino #1054: Unable to start serial monitor

mradamcollins, 2020-06-11:
Issue Type: <b>Bug</b>

Same symptoms as #918 and #851

Click Open Serial Monitor icon (with correct Com port selected
Output shows
[Starting] Opening the serial port - COM3
and then no data.

Other serial monitors correctly show data when connected

Extension version: 0.3.1
VS Code version: Code 1.46.0 (a5d1cc28bb5da32ec67e86cc50f84c67cc690321, 2020-06-10T09:03:20.462Z)
OS version: Windows_NT x64 10.0.18362

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-4702HQ CPU @ 2.20GHz (8 x 2195)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.91GB (9.09GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

Naphier, 2020-06-12:
```
[Starting] Opening the serial port - COM4
[Info] Opened the serial port - COM4
[Starting] Opening the serial port - COM4
[Info] Opened the serial port - COM4
[Done] Closed the serial port 
```
Same here. Starts, Opens, no data. Serial port is tied up as I can't open in any other monitor. But after closing in VS Code, I can open monitor in Arduino IDE and Putty just fine.

hellyzh, 2020-06-15:
Not able to reproduce, what device are you using and is it worked previously?

mradamcollins, 2020-06-15:
Confirmed working with extension version 0.3.0
NOT working in 0.3.1
System specs are in original post
To reproduce: I have an NRF52 transmitting data on serial port (COM3) continuously. Confirmed working by opening Arduino Serial Monitor and observing data.
With 0.3.1 if I click on the "open Serial Monitor"  Icon. I get 
```
[Starting] Opening the serial port - COM3
[Info] Opened the serial port - COM3
```
and then nothing.
With 0.3.0 it opens and streams data

ba05, 2020-07-18:
Same issue with 0.3.1. Other serial monitors work fine.

Gostas, 2020-07-31:
Same issue with 0.3.1. I have to use the serial monitor from arduino ide instead.

yanoom, 2020-07-31:
Same issue.
Sometimes (when switching versions) I get a 25-line dump of serial data, but it stops there (and no prints even when closing/opening serial monitor).

Same behavior with 0.3.0 and 0.3.1 (open serial monitor fails in older versions).

Looks like this:
```
[Starting] Opening the serial port - COM3
[Info] Opened the serial port - COM3
4
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
674
[Done] Closed the serial port 

[Starting] Opening the serial port - COM3
[Info] Opened the serial port - COM3
�[Done] Closed the serial port 

[Starting] Opening the serial port - COM3
[Info] Opened the serial port - COM3
```

Wibbbs, 2020-08-28:
Same issue here. Open the serial port and see this.

[Starting] Opening the serial port - COM3
[Info] Opened the serial port - COM3

I use the following in the setup to initiate 

But then no ouput. Arduino IDE serial monitor works fine on the same sketch.

russelljahn, 2020-09-06:
Same issue here. No output from the VSCode Serial Monitor (even tho my settings and sketch have the same baud rate). Tested with both 9600 and 115200 baud rates.

The serial monitor in the Arduino IDE works fine tho. It wouldn't be a problem to use the Arduino's IDE for the time being, except I'm unable to upload sketches from VSCode while the Arduino IDE is open (permission error).

russelljahn, 2020-09-06:
As a follow-up on the permission error I was encountering when uploading from VSCode with the Arduino IDE open: 

This only happens when the Serial Monitor is open in the Arduino IDE. If you close the Serial Monitor, then you can usually upload firmware from VSCode fine. If you sporadically encounter a permission error afterwards when uploading, restarting the Arduino fixes it for me.

russelljahn, 2020-09-06:
Not ideal - but this is my current workflow which is speedy enough for now:

1. Code firmware and upload in VSCode.
2. Keep the Arduino IDE open.
3. Before reuploading firmware in VSCode, close the Serial Monitor in the Arduino IDE.
4. Upload the firmware in VSCode (**CTRL+ALT+U** on Windows).
5. Reopen the Serial Monitor in the Arduino IDE (**CTRL+SHIFT+M** on Windows)

Even with the extra manual steps, development-wise, this is leagues faster than directly developing in the Arduino IDE.

scarolan, 2020-10-08:
@russelljahn thanks for this - I also cannot get the Serial Monitor working in VSC so the workaround at least gets me unstuck.

I am also on Windows 10.

```
Version: 1.48.2 (user setup)
Commit: a0479759d6e9ea56afa657e454193f72aef85bd0
Date: 2020-08-25T10:13:11.295Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.19041
```

adiazulay, 2020-11-11:
Does v0.3.3 fix this issue?

Gostas, 2020-12-20:
It works for me now on 0.3.4

adiazulay, 2021-01-04:
This appears to be resolved in 0.3.4, please reopen if this is still an issue

closed 2021-01-04 by adiazulay