microsoft/vscode-arduino #1055: Extension not functional in ubuntu 20.04 - cannot use outdated arduino IDE 1.8.6

masynthetic, 2020-06-12:
I am trying to get the Arduino extension to work in my VScode on Ubuntu 20.04 and as far as I can tell at this point it is not possible because according to 

[this ](https://github.com/Microsoft/vscode-arduino/issues/707)and [this](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) 

**we cannot use an IDE beyond 1.8.6 however 1.8.6 does not install correctly on ubuntu 20.04. The IDE does install and (can?) be used but the "arduino" command does not work to launch it which is needed by VSCode. The desktop icon also does not appear. I am not sure what/if else didn't install correctly. If I go to the folder and run ./arduino.sh it opens the IDE. The 1.8.6 IDE is no longer supported.** 

I originally tried using the latest arduino IDE snap and the VScode snap but after discovering the apparent version issue I installed the arduino IDE from tar.xz instead. VScode is still a snap.

hellyzh, 2020-06-23:
Suggest use the latest version of Arduino IDE which currently is 1.8.13

masynthetic, 2020-06-23:
@hellyzh 

[Note: Arduino IDE 1.8.7 has some breaking changes, causing board package and library installation failures.](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino)

hellyzh, 2020-06-24:
Is the issue no board types related to Arduino IDE issue https://github.com/arduino/Arduino/issues/10335 which should be fixed. You can try use the lates arduino IDE again.

hellyzh, 2020-06-24:
#1052 Arduino: Board Manager failed to run caused by Arduino IDE issue

masynthetic, 2020-06-24:
@hellyzh 

still no options found under <select board type> unfortunately

I am using the 1.8.13 arduino snap and have made sure to wipe all traces of the older versions. 

are these configs correct?

"arduino.path": "/snap/bin",
"arduino.commandPath": "arduino",

hellyzh, 2020-06-24:
The Windows Store's version of the Arduino IDE is not supported. Not tested arduino snap. [Arduino IDE](https://downloads.arduino.cc/arduino-1.8.13-linux64.tar.xz) is supported.

masynthetic, 2020-06-25:
> The Windows Store's version of the Arduino IDE is not supported. Not tested arduino snap. [Arduino IDE](https://downloads.arduino.cc/arduino-1.8.13-linux64.tar.xz) is supported.

happy to confirm this one works

closed 2020-06-25 by masynthetic

Barrut827, 2021-04-26:
is it fixed yet?

masynthetic, 2021-04-26:
> is it fixed yet?

It does not work with the snap version of the arduino IDE but works fine if you follow standard install. Vscode is ok as a snap

Barrut827, 2021-04-27:
> > is it fixed yet?
> 
> It does not work with the snap version of the arduino IDE but works fine if you follow standard install. Vscode is ok as a snap

Yep that did the trick, thanks!

Comfort-Twala, 2021-05-31:
> > @hellyzh
> > still no options found under **select board type** unfortunately
> > I am using the 1.8.13 arduino snap and have made sure to wipe all traces of the older versions.
> > are these configs correct?
> > "arduino.path": "/snap/bin",
> > "arduino.commandPath": "arduino",
> 
> Try something like this:
> 
> ```json
> {
>   "arduino.path": "/snap/arduino/56",
>   "arduino.commandPath": "arduino",
> }
> ```
> 
> in `arduino.path` the number 56 may vary.

You're a LEGEND my Gee !!!