microsoft/vscode-arduino #1056: build directory keeps deleted files resulting in "multiple definition" errors during link

phord, 2020-06-14:
When I rename a source file in an arduino project and the "output" folder is set in arduino.json, I get a build error because the old filename is still in the output folder.

1. I created a project with the "Blink.ino" sketch and this arduino.json:

    {
        "board": "esp8266:esp8266:oak",
        "configuration": "xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600",
        "sketch": "Blink.ino",  
        "output": "../test-build"
    }

2. I created a file named `src/foo.cpp` to be included with my project.  It defines a single function.
3. I compiled the project with `Arduino: Verify`.  The build succeeded.
4. I clicked on "bar.cpp" in the File Explorer and pressed F2 to rename it.
5. I changed the name to bar.cpp.
6. I compiled the project with `Arduino: Verify`.  The build failed with "multiple definition.
7. When I look in my `test-build/sketch` directory, I see both `foo.cpp` and `bar.cpp` and related build output.

phord, 2020-06-14:
fwiw - I get the same error if I rename foo.cpp from my shell instead of renaming it in the editor.  But I do not see this error if I delete foo.cpp, either from the shell or from VS Code.

    $ find .                   
    ./.vscode
    ./.vscode/settings.json
    ./.vscode/arduino.json
    ./.vscode/c_cpp_properties.json
    ./src
    ./src/foo.cpp
    ./Blink.ino

    [Starting] Verify sketch - Blink.ino
    Please see the build logs in Output path: /home/phord/test-build
    Picked up JAVA_TOOL_OPTIONS: 
    Loading configuration...
    Initializing packages...
    Preparing boards...
    Verifying...
    Executable segment sizes:
    IROM   : 228508          - code in flash         (default or ICACHE_FLASH_ATTR) 
    IRAM   : 26784   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
    DATA   : 1252  )         - initialized variables (global, static) in RAM/HEAP 
    RODATA : 688   ) / 81920 - constants             (global, static) in RAM/HEAP 
    BSS    : 24880 )         - zeroed variables      (global, static) in RAM/HEAP 
    Sketch uses 257232 bytes (24%) of program storage space. Maximum is 1044464 bytes.
    Global variables use 26820 bytes (32%) of dynamic memory, leaving 55100 bytes for local variables. Maximum is 81920 bytes.
    [Done] Finished verify sketch - Blink.ino

    $ find ../test-build/sketch
    ../test-build/sketch
    ../test-build/sketch/Blink.ino.cpp
    ../test-build/sketch/src
    ../test-build/sketch/src/foo.cpp
    ../test-build/sketch/src/foo.cpp.d
    ../test-build/sketch/src/foo.cpp.o
    ../test-build/sketch/Blink.ino.cpp.d
    ../test-build/sketch/Blink.ino.cpp.o

Now, I just rename foo.cpp to bar.cpp, and the build fails.

    $ mv src/foo.cpp src/bar.cpp
    $ find .                    
    ./.vscode
    ./.vscode/settings.json
    ./.vscode/arduino.json
    ./.vscode/c_cpp_properties.json
    ./src
    ./src/bar.cpp
    ./Blink.ino

    [Starting] Verify sketch - Blink.ino
    Please see the build logs in Output path: /home/phord/test-build
    Picked up JAVA_TOOL_OPTIONS: 
    Loading configuration...
    Initializing packages...
    Preparing boards...
    Verifying...
    /home/phord/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/src/foo.cpp.o: in function `some_function()':
    sketch/src/foo.cpp:1: multiple definition of `some_function()'; sketch/src/bar.cpp.o:sketch/src/bar.cpp:1: first defined here
    collect2: error: ld returned 1 exit status
    exit status 1
    [Error] Exit with code=1

    $ find ../test-build/sketch 
    ../test-build/sketch
    ../test-build/sketch/Blink.ino.cpp
    ../test-build/sketch/src
    ../test-build/sketch/src/foo.cpp
    ../test-build/sketch/src/foo.cpp.d
    ../test-build/sketch/src/foo.cpp.o
    ../test-build/sketch/src/bar.cpp
    ../test-build/sketch/src/bar.cpp.d
    ../test-build/sketch/src/bar.cpp.o
    ../test-build/sketch/Blink.ino.cpp.d
    ../test-build/sketch/Blink.ino.cpp.o

The only way to get things working again is to find my build output folder and manually delete the old sketch files there.

hayden-servaenergy, 2024-01-22:
Still experiencing this issue with v0.6.0. Having a "Build Directory Clean" command would be great for this, at least as a temporary fix.