microsoft/vscode-arduino #1059: PR should either be accepted or rejected, not left in limbo

PeterWone, 2020-06-22:
You processed mine quite quickly, but for some reason some PRs have languished for literally years. In particular I can tell you the author of #767 is running out of patience with bringing his fork up to date for your merging convenience only to be ignored for two years. 

If you don't want the PR then reject it. Don't leave him in limbo.

I know you do your best. You were actually pretty quick when I submitted a fix.

hairlesshobo, 2020-06-23:
Hey, thanks! I appreciate you helping bring attention to this!

PeterWone, 2020-06-23:
No worries. As a matter of interest, I have made a feature request for vscode itself. Do me a favour and take a look https://github.com/microsoft/vscode/issues/100534 see whether you think it's a good idea.

phord, 2020-06-24:
Curiously, #676 was also abandoned for neglect.  Coincidence?

benmcmorran, 2022-01-10:
We have more resources dedicated to this project now, and #1406 outlines our plan going forward. We'll be reviewing new issues and PRs at least weekly. Please feel free to comment on #1406 if you have any other questions or suggestions for how we can improve the governance of this project.

closed 2022-01-10 by benmcmorran