microsoft/vscode-arduino #1060: Serial monitor auto-reconnect is broken after a build failure

phord, 2020-06-24:
Usually the serial monitor is disconnected / switched away when compiling/uploading, and then reconnected / switched back after the board is reset.  That's a nice feature.  On the serial monitor I see this:

    [Done] Closed the serial port 

    [Starting] Opening the serial port - /dev/ttyUSB2
    [Info] Opened the serial port - /dev/ttyUSB2
    Hello, World.

But if a compile fails, this quick-resume feature is lost. When I fix the compile problem and try again, the serial monitor doesn't start up and I miss the first serial messages from my board.

Oftentimes, the code change I made will show me some status at boot on the serial console, and I'd really like to catch this output  on the console without having to manually reset the board to start over.

gcampbell-msft, 2023-02-16:
@phord We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Please let us know if this new implementation fixes your issue, or if this issue still exists.

StefanKellerAC, 2023-03-26:
In v0.6 and v0.6.230316001 it still switches back to the new Serial Monitor tab on not successful compilation.

closed 2023-05-02 by github-actions[bot]