microsoft/vscode-arduino #1061: Serial monitor should not be closed for "build" step during build-upload

phord, 2020-06-24:
The serial monitor is closed when press Ctrl+Alt+U to upload my code.  But if the build fails, no upload occurs and the serial monitor did not need to close.  Also, a long build process interrupts my serial console far too much as it takes longer than the upload itself.

Can you wait until the upload step is about to start before interrupting the serial monitor?

hellyzh, 2020-07-03:
Command `Arduino: Verify` can build without closing serial monitor.

phord, 2020-07-04:
True, but I don't usually run Verify separately for small changes.

maddogjt, 2020-11-01:
The arduino command IDE based command line does not allow for separating the build & upload processes, so the serial montior must be disabled for the duration of the call.

The new arduino-cli tool does support separately building and uploading.  Once this extension is switched over to use arduino-cli, it should be possible to only disable serial monitor during the upload phase.

gcampbell-msft, 2023-02-16:
@phord We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if it still exists!

closed 2023-04-02 by github-actions[bot]