microsoft/vscode-arduino #1064: Board Manager / Library Manager are Empty on VirtualBox / Remote SSH

MitchTalmadge, 2020-06-27:
Howdy. I am trying to use this plugin over remote SSH on a VirtualBox machine because I'd like to develop in Linux even though I run Windows. Unfortunately the board manager and library manager just show empty windows and I cannot find any errors related to this.

Any ideas of where to go from here? I know this may be an odd setup so I appreciate any help.

[image]

- VSCode version: 1.46.1
- OS:
  - Host: Windows 10, v2004
  - Guest: Debian GNU/Linux 10 Buster
- Extension versions:
  - Arduino: 0.3.1
  - Remote SSH: 0.51.0
- Arduino version: 1.8.13

rotogro, 2020-06-28:
Same issue.   All the same versions.     Getting Error: 

/snap/arduino/current/arduino: line 35: java: command not found
[Error] Exit with code=127

Strat00s, 2020-07-06:
Yey, I am not the only one.
I wanted to try the remote development using SSH and everything seems to work fine. Except for Arduino.
And I am also not sure, if it even can work remotely (using my windows PC as I/O and the server as everything else).
PIO also seems to not work as I expected. Hopefully it will get better.

MitchTalmadge, 2020-07-10:
For what it's worth I have switched to WSL (Windows Subsystem for Linux) and it works much better. It does have its own problem though: code formatting takes FOREVER and I have no idea why. But that's the C++ plugin's fault, not this one.

mj2068, 2021-03-27:
> Yey, I am not the only one.
> I wanted to try the remote development using SSH and everything seems to work fine. Except for Arduino.
> And I am also not sure if it even can work remotely (using my windows PC as I/O and the server as everything else).
> PIO also seems to not work as I expected. Hopefully, it will get better.
> 
> P.S.: I am using a headless server, which is (or at least was) an issue for Arduino, so maybe there is still a problem with it.
> 
> Edit: Just as I thought. I can't select serial port, so it is kinda useless for me as of now 😞

I have exactly the same encounters as you, but, tbh, our setups are kind of an odd one... but hopefully, things will get better.

bastibeckr, 2021-12-15:
The empty windows / tabs show a web-view. I tried it locally and it looks good. The URL is something like 
http://localhost:37061/boardmanager...

When opening the dev-tools / inspector in VSCode, indeed a web-view will load and fails (in case of remote). When I try to open this URL on the remote machine (via curl), I get something back.

Of course, when the extension is installed on the remote machine, this will not work out-of-the-box. The localhost:// must be replaced somehow and the http-server must allow external connections. 

This should be the source for the Web-Server: https://github.com/microsoft/vscode-arduino/blob/3bb7b14f1dcf803180e6a16773dccebc9ced6725/src/arduino/localWebServer.ts#L20

I don't know creating a local web server is common practice for extensions and if the problem can be solved by this extension or by remote-ssh extension. There's an option for port-forwarding in the remote-ssh extension, but since the port seems to be random, this won't work.

I don't know the details about VSCode extension development, but maybe this helps. 

[image]

RanzQ, 2021-12-18:
@rotogro Check this config to fix the java not found issue: [#1163](https://github.com/microsoft/vscode-arduino/issues/1163#issuecomment-997155241)

davidcooper1, 2022-02-14:
@MitchTalmadge I think I found the issue. When serving the view to the UI extension, the workspace extension sends a link using `localhost` instead of using `vscode.env.asExternalUri` to automatically port forward from the remote computer. I was personally dealing with this issue trying to remotely develop with my raspberry pi. I'll be reading through the contributing readme and posting a PR hopefully soon.

benmcmorran, 2022-02-17:
Reactivating because the fix is merged to `main` but not yet released.

RuofengX, 2022-05-03:
Same issue running extention v0.4.11, in ssh remote window on raspberryPi.

PR #1459 (2022-02-17): Support Remote Development Board Selection
Currently, the port of the workspace extension on the remote host is not automatically forwarded in the local VS Code instance. This PR uses `vscode.env.asExternalUri` in `arduinoContentProvider` instead of `localhost`, so the port is automatically forwarded. This way, calls to the API don't fail from the local UI. To isolate the issue, I opened up the dev tools for VS Code and manually forwarded the port the Arduino extension was using for the web server. The board view showed up after forwarding, indicating it is not automatically forwarded.

[Micro [... cut]
closed 2022-05-03 by github-actions[bot]