microsoft/vscode-arduino #1067: VSC 1.48.0-insiders vscode-arduino 0.3.1 serial port error

HEMISixty, 2020-07-10:
This morning, I was prompted to update VSC from 1.47.0-insiders to 1.48.0-insiders.  This has apparently broken the serial monitor access for the arduino extension.  The error text is pasted below.  I have reverted back to 1.47.0-insiders, that didn't resolve it.  I updated to 1.48.0 and backed the extention down to 0.3.0, that didn't resolve it.  I'm currently running 1.48.0 and 0.3.1 with the manual workaround of specifying the serial port in the json files.

Failed to open serial port COM8 due to error: + Error: \\?\c:\Users\blahblah\.vscode-insiders\extensions\vsciot-vscode.vscode-arduino-0.3.1\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application.
\\?\c:\Users\blahblah\.vscode-insiders\extensions\vsciot-vscode.vscode-arduino-0.3.1\out\node_modules\usb-detection\build\Release\detection.node

This was last functioning yesterday, prior to the VSC update.  I'll be happy to collect any data or assist with update the extension.

hellyzh, 2020-07-13:
Serial port module is built based on electron 7.x. VSC 1.48.0-insiders updated to electron 8.x which cause serial port not work.

HEMISixty, 2020-07-13:
Alright, I'll watch for an update built on 8.

Thank you...

manos-msdn, 2020-09-20:
I confirm this issue with the latest vs code, managed to workaround this by downgrading to 1.47.3 following the instructions here: https://stackoverflow.com/questions/49346733/how-to-downgrade-vscode (remember to disable auto update first)

closed 2020-10-22 by HEMISixty