microsoft/vscode-arduino #1068: How to add boards with out  Additional Board Manager URL?

tumuyan, 2020-07-15:
I'm useing a ESP32S2 dev board. Now it is not add to any Additional Board Manager URL .
Download files from esp32s2 branch and unzip it to the "arduino-nightly\hardware" folder, then the arduino IDE could show the board . (refer. https://esp32.com/viewtopic.php?f=2&t=15603&p=59980&hilit=arduino#p59822)
But the vscode could not find these boards. also I have not find out a json file like a additional board config in my pc.
So how to add boards with out  Additional Board Manager URL?

robotdad, 2020-10-30:
Add it to arduino.additionalUrls in settings.json, e.g as here (but use the url you need).:
"arduino.additionalUrls": [
        "https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json",
        "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json",
        "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
    ],

tumuyan, 2020-11-03:
> Add it to arduino.additionalUrls in settings.json, e.g as here (but use the url you need).:
> "arduino.additionalUrls": [ [...]

but the problem is idon't have the json file's url, i only download the sdk directly and then the arduino ide could work with the sdk.

adiazulay, 2020-12-16:
To add the ESP32S2 you have to overwrite some files in the standard ESP32 board package.

Steps:
1. Install the ESP32 board package normally
2. [Download the ESP32S2 branch](https://github.com/espressif/arduino-esp32/archive/esp32s2.zip) as ZIP from espressif 
3. From the Zipped folder copy ```cores | libraries | tools | variants | boards.txt | platform.txt | programmers.txt``` into the ESP32 package overwriting the existing files (you can make a copy of the files in there as a backup). The path to write the files in is ```C:\Users\[your-username]\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4```

closed 2020-12-16 by adiazulay