microsoft/vscode-arduino #1072: IntelliSense shows multiple errors but program compiles and runs fine on Arduino Uno

Gostas, 2020-07-16:
I am making a program for my Arduino Uno using the Arduino extension in vs code. Although my program compiles, uploads and runs just fine, I get multiple errors and I can't find why. From my understanding, there seem to be a few bugs with IntelliSense and this should be one of them.

c_cpp_properties.json:
[image]

arduino.json:
[image]

Example of errors:
[image]
[image]

Any help appreciated!

hellyzh, 2020-07-17:
You can try to change the default "compilerPath". It is in the compile log which should be `"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++"`.

Gostas, 2020-07-17:
That helped, thank you!
However, I now get "IntelliSense mode msvc-x64 is incompatible with compiler path."
What should the IntelliSense mode be?

HEMISixty, 2020-07-18:
I followed the suggestion and this is what eliminates both problems:

            "intelliSenseMode": "gcc-x64",
            "compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-g++.exe",

Gostas, 2020-07-20:
Yup that works, cheers!

closed 2020-07-20 by Gostas