microsoft/vscode-arduino #1074: Was compiled against a different Node.js version using NODE_MODULE_VERSION 57. This version of Node.js requires NODE_MODULE_VERSION 76.

matthew-william-lock, 2020-07-18:
I have tried to remove and use npm install to fix this issue but again and again I get the same error when trying to open the serial port

``
Failed to open serial port /dev/ttyUSB0 due to error: + Error: The module '/home/matthew/.vscode-insiders/extensions/vsciot-vscode.vscode-arduino-0.3.1/out/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 76. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

I was having this issue before, and it disappeared somewhat by itself. It has now reemerged and I have spent hours trying to fix it

HEMISixty, 2020-07-18:
You're running into the same Electon issue I ran into.  What is your Electron version installed?  Is it still 7 or 8?  I'm thinking it's 8 and I'm guessing you're doing something with Arduino code.  I could be way off...who knows...but, this is what I ran into last week...

hellyzh, 2020-07-29:
#1067 VSC 1.48.0-insiders vscode-arduino 0.3.1 serial port error

trullock, 2020-10-19:
I have this since upgrading my Node to 14.14.0

>>>Failed to open serial port COM6 due to error: + Error: The module '\\?\c:\Users\Andrew Bullock\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 75. This version of Node.js requires NODE_MODULE_VERSION 80. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

robotdad, 2020-10-30:
Resolved in release 0.3.3

closed 2020-10-30 by robotdad

caesar, 2021-02-05:
This is happening again today after upgrading to VS Code 1.53.0.

Error is
```Failed to open serial port /dev/tty.usbserial-1460 due to error: + Error: The module '/Users/caesar/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.4/out/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 80. This version of Node.js requires NODE_MODULE_VERSION 85. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).```

Uninstalling and reinstalling the extension doesn't help.

nik1a, 2021-02-05:
Same error here.

shoop, 2021-02-06:
Installing the latest release from VSIX works for me, https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.5

Manguil, 2021-02-13:
@shoop thanks for you. Your method works for me

chovanj, 2021-02-13:
Upgrading from v0.3.4 to v0.3.5 helped me.

scaleb53, 2021-02-18:
> Upgrading from v0.3.4 to v0.3.5 helped me.

@chovanj How did you upgrade? It says the latest version is 3.4 in VS extensions. Is there another way?

chovanj, 2021-02-18:
> > Upgrading from v0.3.4 to v0.3.5 helped me.
> 
> @chovanj How did you upgrade? It says the latest version is 3.4 in VS extensions. Is there another way?

You have to download it from here: https://github.com/microsoft/vscode-arduino/releases and manually install to your VS Code because it's not in the extensions repository yet.

trullock, 2021-05-08:
Same as ever, uninstall VSC, reformat windows, buy a new pc, have a brain transplant, move to another planet and then slide to a parallel universe.

Alternatively wait until next month when the authors republish this terribly brittle extension.

Go to settings in VSC and disable auto update, than revert to the April release and resume doing work

membiblio, 2021-05-08:
Andrew - I followed your instructions to the letter and even after arriving in the parallel universe the serial console refuses to open/work. Strangely it compiles and uploads using the said serial port but no console for debugging joy. The fake steaks in this universe are delicious and they have hovercars! :)

Thank you for your advice - I even tried rolling back one at a time until I hit Arduino Pluging 0.3.0 from a year ago and similar failure message including no serial console joy.  As you said I will wait until one of the releases works and then pin it in place and not update it.

This whole Arduino under Visual Studio looks quit appealing and I applaud Microsoft for their hard work and I'll stop there. :)

Thank you again Andrew you at least made me smile :)

trullock, 2021-05-08:
Not at my computer to cite the exact versions, but the latest Arduino extension works with the previous VSC version from April

membiblio, 2021-05-08:
Andrew Bravo!  Success!  Works as you would expect it to :)

Visual Studio Code 1.55.2 from April 21 which can be downloaded here (to assist others):
https://code.visualstudio.com/updates/v1_55

And the Visual Studio Code Extension: Arduino for Visual Studio Code 0.4.2 
Which was already installed in the ide as it remembered it even after uninstalling and installing 1.55.2

Thank you Andrew you are a Superstar!  :)