microsoft/vscode-arduino #1075: include library zip

jolugama, 2020-07-19:


hellyzh, 2020-07-22:
Please try to extract the zip file to C:\Users\[user]\Documents\Arduino\libraries, and run `"Arduino: Library Manager"` to include the library to the example sketch. Also you can check the guide for [writing a library for Arduino](https://www.arduino.cc/en/Hacking/libraryTutorial).

jolugama, 2020-07-22:
It`s working!! in c_cpp_properties.json, includePath, i add this : "/home/MYUSER**/snap/arduino/current/Arduino",

This directions is the same like in arduino: file, preferencies, proyect location. 

Then,  right button on mouse in library in visual code, add library.     

Thanks!!!!

closed 2020-07-22 by jolugama

olawalejuwonm, 2024-06-10:
> It`s working!! in c_cpp_properties.json, includePath, i add this : "/home/MYUSER**/snap/arduino/current/Arduino",
> 
> This directions is the same like in arduino: file, preferencies, proyect location.
> 
> Then, right button on mouse in library in visual code, add library.
> 
> Thanks!!!!

Please I do not understand this.

How can I make this work for me?