microsoft/vscode-arduino #1076: Build failed on Ubuntu 18.04 / WSL 20.04

Meetem, 2020-07-19:
Hello, I'am trying to implement some minor functionality to vscode-arduino plugin, but it won't build without any changes.
master/develop branches was tried.

Installation process:
1. Cloned repo
2. ``npm install``
3. ``npm install -g gulp``
4. ``F5`` in vscode for debug.

Gulp build log:
```
[00:46:08] Using gulpfile ~/vscode-arduino/gulpfile.js
[00:46:08] Starting 'build'...
[00:46:08] Starting 'clean'...
[00:46:08] Finished 'clean' after 7.66 ms
[00:46:08] Starting 'ts-compile'...
[00:46:14] Finished 'ts-compile' after 5.98 s
[00:46:14] Starting 'html-webpack'...
[00:46:14] 'html-webpack' errored after 1.2 ms
[00:46:14] Error: Cannot find module 'html-webpack-plugin'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/alex/vscode-arduino/src/views/webpack.config.js:2:27)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
[00:46:14] 'build' errored after 6 s
```

Same output on Ubuntu 18.04 or WSL Ubuntu 20.04.
For windows: npm install failing while building usb-detection library.

hellyzh, 2020-07-22:
What is your version of nodejs, 10.x should work.

MrThanlon, 2020-07-23:
> What is your version of nodejs, 10.x should work.

10.22.0, same issue.

Meetem, 2020-07-23:
> What is your version of nodejs, 10.x should work.

It's 10.19.0

hellyzh, 2020-07-24:
Please try to refer[ travis ci](https://github.com/microsoft/vscode-arduino/blob/master/.travis.yml#L40) to prepare the environment.