microsoft/vscode-arduino #1077: Wrong board detected from vscode

florianvoss99, 2020-07-20:
Hello,
I connected an ESP32 board to my device but vscode detects an ESP82. Have tried to search manually in the board manager with same results, it would only show me ESP82 options. That made me think that the samples for an esp82 also work for an esp32 board but sadly thats not the case. Any code I try to send, won't arrive. 
I am very thankful for any solutions, greetings everyone.

hellyzh, 2020-07-22:
You can check the [example](https://docs.microsoft.com/en-us/samples/azure-samples/esp32-iot-devkit-get-started/sample/) for ESP32 which includes its board config.

florianvoss99, 2020-07-22:
No I cannot, since the ESP32 won't even be detected from vsc. Not the board config and no examples or samples or sketches in general. Like I said the board is not even being detected from vsc, nor from the Arduino IDE application. I will probably consult someone from the company, thx for trying to help.

closed 2020-07-22 by florianvoss99