microsoft/vscode-arduino #1082: After 4-5 uploads the application crashes

mightycoco, 2020-08-04:
ERR An unknown error occurred. Please consult the log for more details.

  | _onExtensionHostCrashed | @ | abstractExtensionService.ts:166
-- | -- | -- | --
  | _onExtensionHostCrashed | @ | extensionService.ts:417
  | _onExtensionHostCrashOrExit | @ | abstractExtensionService.ts:158
  | (anonymous) | @ | abstractExtensionService.ts:148
  | fire | @ | event.ts:587
  | _onExtHostProcessExit | @ | localProcessExtensionHost.ts:502
  | (anonymous) | @ | localProcessExtensionHost.ts:249
  | emit | @ | events.js:203
  | ChildProcess._handle.onexit | @ | internal/child_process.js:272

Installed Arduino IDE 1.8.12
I don't know, how to get more detailed logging except the output from Developer Tools

mightycoco, 2020-08-04:
Found this one

[2020-08-04 11:38:12.022] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:38:12.022] [exthost] [error] undefined
[2020-08-04 11:38:17.092] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:38:17.092] [exthost] [error] undefined
[2020-08-04 11:38:20.385] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:38:20.385] [exthost] [error] undefined
[2020-08-04 11:39:03.431] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:39:03.431] [exthost] [error] undefined
[2020-08-04 11:39:11.687] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:39:11.687] [exthost] [error] undefined
[2020-08-04 11:39:44.793] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:39:44.793] [exthost] [error] undefined
[2020-08-04 11:39:56.869] [exthost] [warning] ms-vscode.cpptools - Code actions of kind 'quickfix 'requested but returned code action does not have a 'kind'. Code action will be dropped. Please set 'CodeAction.kind'.
[2020-08-04 11:39:56.869] [exthost] [warning] ms-vscode.cpptools - Code actions of kind 'quickfix 'requested but returned code action does not have a 'kind'. Code action will be dropped. Please set 'CodeAction.kind'.
[2020-08-04 11:42:45.352] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:42:45.356] [exthost] [error] Error: DisposableStore is missing...
	at U._convertCompletionItem (c:\Users\xxx\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:679:652)
	at U.resolveCompletionItem (c:\Users\xxx\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:679:419)
[2020-08-04 11:42:45.785] [exthost] [error] [ms-vscode.cpptools] provider FAILED
[2020-08-04 11:42:45.785] [exthost] [error] undefined

Don't know, if this helps...

hellyzh, 2020-08-21:
[ms-vscode.cpptools] c/c++ intellisense is an extension dependency of arduino extension. The crash should be not related to Arduino extension, reopen vscode may fix the issue.

robotdad, 2020-10-30:
Yes, that looks like the C++ extension which this one depends on. Please reactivate if it happens again and I can help route it to the right place.

closed 2020-10-30 by robotdad