microsoft/vscode-arduino #1083: How can I add new board?

crystalcave, 2020-08-08:
Usually I'm using Teensy board, that downloaded by [Tensiduino](https://www.pjrc.com/teensy/teensyduino.html).
Actually I'm developing by Visual Studio Code with this Extension, but for compile, I have to open Arduino IDE.
This one is necessary because, in board list (using Arduino Extension) Tensiduino not appears.

I think that board list of Arduino Extension not based with the same approach used by Arduino IDE to retrieve all boards.

How can I show Teensy board into VS Code Arduino Extension?

robotdad, 2020-10-30:
Did you try adding the url to the teensy board package to arduino.additionalUrls in the settings.json?

firelizzard18, 2021-02-20:
@robotdad What URL? I don't see any explanation of how the URL functions, so I have no idea how to come up with a URL that will add Teensy boards.

jmulvey123, 2023-03-25:
The additional URL you need to specify for Teensy is https://www.pjrc.com/teensy/package_teensy_index.json