microsoft/vscode-arduino #1086: Include file not found in browse.path. [1,1]

soham2002, 2020-08-20:
I had just started learning C programming for my first semester. But VS Code is showing some problems...
Is there a way to fix this? I had installed the **C/C++** and the **Code Runner** extensions, but still, it's not compiling  :(

[image]

hellyzh, 2020-08-21:
Please follow the [guide](https://code.visualstudio.com/docs/cpp/customize-default-settings-cpp) to set up the config file c_cpp_properties.json, and open issue in the repo of [c/c++ intellisense](https://github.com/Microsoft/vscode-cpptools).

adiazulay, 2020-11-17:
This appears to be a c/c++ issue, you can file in the c/c++ extension repo: https://github.com/microsoft/vscode-cpptools/issues

closed 2020-11-17 by adiazulay