microsoft/vscode-arduino #1094: Syntax highlighting and #ifdef

didiergm, 2020-09-02:
VSCode 1.48.2 / Arduino extension 0.3.2

When using #define to use different chunks of code with #ifdef .. #endif the portions of code /excluded/ used to be greyed out by the syntax highlighter.

it does not seem to be the case anymore not sure whether if it comes from a VSCode update or the extension update.

Hope this is clear as English is not my main language

hellyzh, 2020-09-03:
The language intelliSense function is provided by the extension dependency of C/C++. I checked that the excluded portions of code are greyed out if installed the latest version 0.29.0 of C/C++ extension.
[image]

didiergm, 2020-09-03:
Just checked I am on 0.30.0-insider4, tried to downgrade to 0-29-0 but
VSCode reloaded 0.30.0-insider4 immediately, not sure what to do ...

hellyzh, 2020-09-03:
Please try to "Disable Auto Updating Extensions" according to https://code.visualstudio.com/docs/editor/extension-gallery.

didiergm, 2020-09-05:
It did work & indeed v 0.29.0 of the extension displays the desired
behaviour

adiazulay, 2020-11-17:
this appears to be resolved in c/c++ extension. Please reopen if this is still an issue.

closed 2020-11-17 by adiazulay