microsoft/vscode-arduino #1096: Output panel encoding error

Tnze, 2020-09-10:
When I click the Verify button:
It shows
```
椤圭洰浣跨敤浜� 206423 瀛楄妭锛屽崰鐢ㄤ簡 (15%) 绋嬪簭瀛樺偍绌洪棿銆傛渶澶т负 1310720 瀛楄妭銆�
鍏ㄥ眬鍙橀噺浣跨敤浜�14556瀛楄妭锛�(4%)鐨勫姩鎬佸唴瀛橈紝浣欑暀313124瀛楄妭灞€閮ㄥ彉閲忋€傛渶澶т负327680瀛楄妭銆�
```
which should be
```
项目使用了 206423 字节，占用了 (15%) 程序存储空间。最大为 1310720 字节。
全局变量使用了14556字节，(4%)的动态内存，余留313124字节局部变量。最大为327680字节。
```
Change the language setting in Arduino into (Chinese (China)) could recurring this bug:
[image]
[image]

It is duplicated with #958 but I don't think change the global system setting is a good idea. 
In my situation, it crashes another software on my computer, which is something I can't accept.

I think there should be a encoding setting item in the vscode setting page.

Tnze, 2020-09-10:
It looks like Arduino passed in UTF-8 text, but the output page of vscode tries to interpret it in GB18030 or GB2312 or GBK encoding.

mzhboy, 2020-11-30:
duplicated : #875

Tnze, 2020-11-30:
Is there a perfect solution?

GF-Huang, 2021-02-21:
Any workarounds?

Tnze, 2021-02-21:
It looks like they're not going to fix this bug.

GF-Huang, 2021-02-21:
Sad, so keep issue opening.

benmcmorran, 2022-01-13:
This issue should be fixed by PR #1271 which has merged and will be included in the next release of this extension. If you'd like to test the changes before they are released, you can [download and install the build artifact](https://github.com/microsoft/vscode-arduino/suites/4930939531/artifacts/142502560).

closed 2022-01-15 by github-actions[bot]