microsoft/vscode-arduino #1097: Serial monitor not works after VSCode updated to 1.49.0

hellyzh, 2020-09-11:
Serial monitor not works after VSCode updated to 1.49.0 with Electron 9.2.1 . Error:
``
Failed to open serial port COM3 due to error: + Error: The module '\\?\c:\Users\***\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 75. This version of Node.js requires
NODE_MODULE_VERSION 80. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

Enchan1207, 2020-09-11:
I face same issue too.  
The port selection screen is not displayed even if I click `<Select Serial Port>`, so I can't operate serial monitor.  
(The same applies when execute using command palette.)

rrakso, 2020-09-11:
Yup! Me too! :<

jPerotto, 2020-09-11:
UP, me too :/

HEMISixty, 2020-09-11:
If this is updated, #1067 goes away as well...

vfranchi, 2020-09-11:
I'm having the same issue, any workaround for this?

jPerotto, 2020-09-11:
Yes, I downloaded version 1.48.2, and it worked again.
https://code.visualstudio.com/updates/v1_48

> I'm having the same issue, any workaround for this?

nikhillokeshc, 2020-09-11:
> Yes, I downloaded version 1.48.2, and it worked again.
> https://code.visualstudio.com/updates/v1_48
> 
> > I'm having the same issue, any workaround for this?

I was having the same issue. This version of vscode is working for me too.

Josverl, 2020-09-13:
Same problem,
the underlying issue is that NodeJS makes it hard to interface with native hardware interfaces. 
for this node uses something called native bindings , which have a tendency to break at each and every update.
until NodeJS has a better strategy for this , every extension developer will need to de the work to include updates for a number of future versions on NodeJS.
I have solved this for the pymakr extension which used to have the same problem with serial ports

it would be nice if the VSCode platform included at least one serialport native binding that all extension makers could use to avoid everone needing to setup a CI/CD just to avoid this breaking.

thinkOfaNumber, 2020-09-14:
See #922 for more history on this. The only solution is to go back to Code 1.48.2 until an updated extension version is released.

maxymoo22, 2020-09-15:
Got the same issue

ETLin, 2020-09-16:
i got same.

ETLin, 2020-09-16:
> Yes, I downloaded version 1.48.2, and it worked again.
> https://code.visualstudio.com/updates/v1_48
> 
> > I'm having the same issue, any workaround for this?

worked for me too.

gelanchez, 2020-09-19:
Same issue under GNU/Linux (Kubuntu 18.04)

klanger59, 2020-09-20:
Same issue under Windows 10 for VSCode 1.49.1

PeterWone, 2020-09-20:
> Same problem,
> the underlying issue is that NodeJS makes it hard to interface with native hardware interfaces such as COMX: or /dev/usb123
> for this node uses something called native bindings , which have a tendency to break at each and every update.
> until NodeJS has a better strategy for this , every extension developer will need to de the work to include updates for a number of future versions on NodeJS.
> I have solved this for the pymakr extension which used to have the same problem with serial ports
> 
> it would be nice if the VSCode platform included at least one serialport native binding that all extension makers could use to avoid everone needing to setup a CI/CD just to avoid this breaking.

@Josverl while I agree that the right place to fix this is in VSCode, how did you fix this? 

 * Did you use N-API?
 * Is there a particular commit I should look at?

Josverl, 2020-09-28:
@PeterWone 

the approach was to : 
1.  download the native bindings for upcoming VSCode versions --> Electron Versions --> native Bindings 
 - this required some creative manners to get that information 
- then additional creativity to download the native bindings 
- then even more to find that some of the pre-comibles offered were actually broken
2 - figure out a way to store multipole versions / platforms of ntive bindings that can be located by the load-libraries used by serialport 
2 create test projects to run against electron version N  on Platform X , with bitness Y , with the downloaded binaries 
3 run CI/CD [Nightly / on Commit / PR ] using GitHub actions that flags if upcoming versions ( ie Insiders ) will be broken 
4 [**did not Implement**] create a automated PR with new native bindings as they are released 
5 requires manual action on failed CI/CD Builds 

something like the above would be needed  for each Project 
Please refer to: 
https://github.com/pycom/pymakr-vsc/pull/59
( i learned more than i wanted to know when figuring out a workaround for something that i assumed was a trivial fix ....) 

but as you can see , also pymakr had the same issue as there was still a manual step to watch the CI/CD against for the insiders build.
and even including Serialport in VSCode will not solve the problem , only shift it to a place where it can be managed, but actually it should not need to be managed/contained, it should be prevented. 

I agree that N-API would be much better, and shift the responsibility to the creators of library , but [the suggestion for that](https://github.com/serialport/node-serialport/issues/1186) seems 3 year old with no followup that i could find.

HeAtNet, 2020-09-29:
Same issue under Windows 10 for VSCode 1.49.2.
Also cannot select other serial port.
Uploading works fine.

PeterWone, 2020-09-30:
Yes, I reached similar conclusions.

What I have done instead is create a whole new extension that integrates the new Arduino-CLI.

This is currently an incomplete prototype but progress is rapid. So far I have support for

  *   Choose board (and by implication toolchain, header files and compatible programmers)
  *   Choose programming method (a port or a compatible programmer)
  *   Choose libraries (which causes them to be installed if necessary)
I also have working proofs of concept for

  *   Compile
  *   Flash board
  *   Flash bootloader
But these use hardcoded values and must be reworked to use selections after I finish writing the environment management aspects. At the moment I’m working on managing compiler paths based on library selections.

pycom/pymakr-vsc#59<https://github.com/pycom/pymakr-vsc/pull/59>

gren236, 2020-10-10:
Same problem here. Looking forward to a fix!

kseyhan, 2020-10-22:
check the solution proposed here https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-714255377 if have a proper build environment on your machine.

Josverl, 2020-10-22:
I think the point is not that that (rebuilding the native module) **can** or **cannot** be done by some users, on some machines.

In my view  a user should not _need_ to compile from source to use a published vscode add-in .
And specifically if that in fact means that **all users** on all **platforms** and all **CPU architectures** and all **bitnesses** will need to rebuild from source every time that the electron environment hits an upgrade bump..

that is similar to to getting a car (for free or paid) only to find that you need to rebuild the clutch every 5 months. 
the reasoning that the parts are is included , the tools are free to use and the manual is on the internet does not address the principle point.

The current serial implementation is using the old-tech NAN , the requires a rebuild after very bump in the road .
the future stable solutions are : 
- [N-API](https://nodejs.org/api/n-api.html#n_api_n_api) : for native modules in C 
- [node-addon-api](https://github.com/nodejs/node-addon-api) : for native modules written in C++

so if you care , please ask the serialport maintainers to support one of these APIs , or if you have skills in C / C++ help pitch in the create or test that
https://github.com/serialport/node-serialport/search?q=%22N-API%22&type=issues

kseyhan, 2020-10-22:
@Josverl i'm not realy deep enough into the architecture of the vscode extensions but normaly it should be sufficient to simply distribute the platform dependend extensions from a central point like serialport is doing but i dont know if this is done with vscode extensions at all.

But to be true i'm neither going to run after the maintainers from this scary piece of software not i want to be integrated into it at all. My way would be to find a complete different approach like maybe a better library to use ( unfortunaltely i'm not deep enough in nodejs that i would know a solution right now) or simply going @PeterWone 's approach by just integrating the arduino cli, shelling out all those problems and having a much better solution in the end.

hasantezcan, 2020-10-23:
I just update the engines section in **`package.json`** and I did not see no more errors
```js
//in package.json
	"engines": {
		"vscode": "^1.5.0"
	},
```
update like that 
```js
	"engines": {
		"vscode": "^1.34.0"
	}, 
```

PeterWone, 2020-10-25:
> If the arduino ide manages to have a way to communicate with their drivers plattform independend from java it shouldnt be to hard to port that over to nodejs.

It has conditional compiles and platform specific code for calls into native code. Porting their method to Node will end up with exactly the same problems you are currently experiencing. That is why I chose the approach I am using.

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

robotdad, 2020-10-30:
Resolved in release 0.3.3

closed 2020-10-30 by robotdad

luanpham107, 2020-12-05:
Upgraded to version 1.51.1. The Problem was fixed.

jalbarra, 2021-02-06:
Hi,
I have the same problem, when I want open the serial monitor this is the message, and I have the last vscode version,

"Failed to open serial port COM8 due to error: + Error: The module '\\?\c:\Users\Jalbarra\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 80. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`)."

I have this versions,
Version: 1.53.0 (user setup)
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T20:36:38.611Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19041

pawelkedra, 2021-02-07:
Problem still occurs on 1.53.0 (Mac OS):

```
Failed to open serial port /dev/tty.usbmodem14101 due to error: + Error: The module '/Users/pawelkedra/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.4/out/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 80. This version of Node.js requires NODE_MODULE_VERSION 85. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).
```

Version: 1.53.0
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T16:01:04.046Z (3 days ago)
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 20.3.0

matburnham, 2021-02-11:
Just had the same problem with 1.53.2 on Windows.

[image]

Version: 1.53.2 (user setup)
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:04.245Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19041

Benargee, 2021-02-15:
Also having the same issue 
```
Failed to open serial port COM5 due to error: + Error: \\?\c:\Users\benrg\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application.
\\?\c:\Users\benrg\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node
```
```
Version: 1.53.2 (user setup)
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:04.245Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042
```

cssinate, 2021-02-15:
@Benargee - https://github.com/microsoft/vscode-arduino/issues/1198#issuecomment-774746151 works

HowardHonig, 2021-08-06:
Same problem as others in this thread on opening the serial monitor.  I succumbed to a forced upgrade.  I am getting so sick of all these frigin bugs in the Arduino extension.