microsoft/vscode-arduino #1098: Unable to open com port

tranvan110, 2020-09-12:
Issue Type: <b>Bug</b>

Failed to open serial port COM15 due to error: + Error: \\?\d:\PortableApp\VSCode\data\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application.
\\?\d:\PortableApp\VSCode\data\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node

Extension version: 0.3.2
VS Code version: Code 1.49.0 (e790b931385d72cf5669fcefc51cdf65990efa5d, 2020-09-10T13:22:08.892Z)
OS version: Windows_NT x64 6.1.7601

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-3340M CPU @ 2.70GHz (4 x 2691)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.91GB (3.53GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

nullmind, 2020-09-16:
Have same issue. No problem using the port to upload to device.
Extension version 0.3.2

Version: 1.49.0 (system setup)
Commit: e790b931385d72cf5669fcefc51cdf65990efa5d
Date: 2020-09-10T13:22:08.892Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18363

Forailan, 2020-09-16:
Have same issue.

Version: 1.49.0 (user setup)
Commit: e790b931385d72cf5669fcefc51cdf65990efa5d
Date: 2020-09-10T13:22:08.892Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18362

Extension version: 0.3.2

rroessler, 2020-09-17:
Same issue. Can upload and verify, but cannot change serial port/nor see serial port menus. Additionally cannot access serial monitor as a result of this.

Version: 1.49.0
Commit: e790b931385d72cf5669fcefc51cdf65990efa5d
Date: 2020-09-10T17:39:53.251Z (6 days ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Darwin x64 19.6.0

[image]

MaxBoyman, 2020-09-17:
Same Problem here. VS Code 1.49.0 , Arduino extension 0.3.2.
cannot choose serial port from bottom-bar, cannot verify sketch, cannot upload sketch, cannot open serial monitor.
everything sucks.
Version: 1.49.0 (user setup)
Commit: e790b931385d72cf5669fcefc51cdf65990efa5d
Date: 2020-09-10T13:22:08.892Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

Ghakamo, 2020-09-17:
Having same issue, notable after a vscode update and C/C++ extension update.  Uploads to device fine, only change was the update

ciscojmg, 2020-09-17:
Tengo el mismo problema con el VSCode, puedo **verificar** los programas, puede **subirlo**, pero los puertos COM no los reconoces de forma automatica y tampoco puedo **debugguear** los programa.

ciscojmg, 2020-09-17:
[image]

jgusr, 2020-09-17:
Tengo el mismo problema en Ubuntu
[image]

johanm57, 2020-09-17:
Tengo el mismo problema: No puedo configurar de forma automática el puerto COM y tampoco puedo usar el debug.. Si me funciona la compilación.

[image]

ciscojmg, 2020-09-18:
El problema se corrigio descargando la [version ](https://code.visualstudio.com/updates/v1_48) **VScode**

Version: 1.48.2 (system setup)
Commit: a0479759d6e9ea56afa657e454193f72aef85bd0
Date: 2020-08-25T10:13:11.295Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18362

dotponder, 2020-09-20:
Same issue.

cdelcristo, 2020-09-20:
Same Issue

kbeaugrand, 2020-09-21:
Same issue

simonl65, 2020-09-21:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```

simonl65, 2020-09-21:
Related issues: #922 #1105 #1104 #1101 #1099 #1098

Roberto6969, 2020-09-21:
C'mon guys ... Who will solve this banal error? Don't release new version with so banal error. Visual studio code and Arduino extension is totally unusable with such an error. I understand, bugs could happen, but not such a stupid one. Does Micorsoft has fingers in this errror?

iqapps, 2020-09-22:
Me too.

[image]

Downgrading to previous version dont work anymore....

fidelbrea, 2020-10-01:
Downgrade. This is a solution. I fixed it by downgrading up to 1.48.2.
I hope you find this useful to fix your own situations.

mattkemp, 2020-10-12:
+1 @iqapps 
Working now.
Downgraded to Visual Code 1.48.2 and Arduino extension 0.3.0
Have also turned off automatic updates on Visual Code, just not good enough to break stuff like this with updates.

iqapps, 2020-10-12:
This is still a very serious problem for me.

Some of us developers are stuck on old code here.

What is keeping this issue from being solved ?

MaxBoyman, 2020-10-12:
This issue is being alive for over than a month and no body's gonna send us at least a report from what's going on. Is there anybody in microsoft to fix this? or arduino extension from microsoft is like other microsoft garage projects?

trullock, 2020-10-19:
I've caused this by updating my version of Node.

But I now cant revert this by downgrading Node....

simonl65, 2020-10-19:
@trullock: do you mean that *you* are the cause of this issue(!), or you now have the issue?

If the latter then you'll need to downgrade to VSCode 1.48.2 (and maybe also Arduino extension 0.3.0)

trullock, 2020-10-19:
I mean I caused it to myself :D

Or at least I think thats whats caused it. Guess I'll downgrade if thats a known workaround

Seefer, 2020-10-20:
Just got an MXChip and trying to follow the IoT DevKit Getting Started tutorial on Docs site and having issues with this silly COM port bug mentioned here. This is an abysmal user experience for beginners. I've decided to try PlatformIO and the tutorials there as many tell me that extension has far less pain-points getting in the way of learning this shizzle 👎

kseyhan, 2020-10-22:
check the solution proposed here https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-714255377 if have a proper build environment on your machine.

WenukaB, 2020-11-01:
This can probably be due to an issue with Arduino Installation. I had the same issue. I uninstalled the Arduino existing installation and installed the latest version (1.8.13) it fixed the issue.

simonl65, 2020-11-01:
Arduino plug-in 0.3.3 fixes this issue

MaxBoyman, 2020-11-02:
the problem still persistent even after updating to 0.3.3

mattkemp, 2020-11-02:
Arduino extension 0.3.3 and Visual Code 1.50.1 is working for me.

adiazulay, 2020-11-17:
fixed in 0.3.3

closed 2020-11-17 by adiazulay

kyro-dev, 2021-02-16:
Not fixed for me. I still have the bug. Downgrading to 0.3.3 didn't help.

MaxBoyman, 2021-02-16:
me too....
nothing helped. downgrading, reinstalling, reinstalling the os, and also vscode itself, nothing.

adiazulay, 2021-02-16:
please see #1198