microsoft/vscode-arduino #1099: Cannot select serial port (VS Code, Arduino Extension)

0923416, 2020-09-12:
Issue Type: <b>Bug</b>

Installed latest version of visual studio code, then installed latest version of Arduino extension. I plugged in the Arduino Uno and tried to select the serial port which it is connected to. However I can't open the menu to select a serial port. Neither the F1 terminal or the button in the blue bar works.

Extension version: 0.3.2
VS Code version: Code 1.49.0 (e790b931385d72cf5669fcefc51cdf65990efa5d, 2020-09-10T13:22:08.892Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|AMD Ryzen 7 3700X 8-Core Processor              (16 x 3600)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.95GB (10.68GB free)|
|Process Argv|--crash-reporter-id ccc4ce18-4fca-4b09-b7a1-1a8b9a2e35f3|
|Screen Reader|no|
|VM|0%|
</details>

CD10h, 2020-09-14:
Same issue here. At the moment the workaround is to manually change it in .vscode\arduino.json.

BananaFructa, 2020-09-14:
Same issue here as well.

rotogro, 2020-09-14:
Same issue here.   Have to roll back VS Code one version but snap keeps upgrading it by the end of the day.   Royal pain.

Roberto6969, 2020-09-14:
Same issue here. Real PITA, cannot open serial port at all. Reinstall of Arduino Extension doesn't help

[image]

yoav79, 2020-09-14:
Same issue here as well.

W3AXL, 2020-09-15:
Same behavior seen here on Win10
VSCode 1.49.0
Extension 0.3.2

Clicking on <Select Serial Port> or running the command from F1 does nothing.

sanderao, 2020-09-15:
same issue with below  set up
VSCode Version: 1.49.0 
Visual Studio Code extension for Arduino: 0.3.2
OS: Windows_NT x64 10.0.17763

thelensmanize, 2020-09-15:
I have the same issue on mac

maxymoo22, 2020-09-15:
Damn I cant open the port selection either

khiemnt00, 2020-09-16:
Same issue here !!!!!

menuRivera, 2020-09-16:
Same issue ma fellas

adiazulay, 2020-09-17:
I'm having the same issue as well

Elektrik1, 2020-09-18:
Same issue here!

ciscojmg, 2020-09-18:
En este issues tambien se refleja el mismo problema:
https://github.com/microsoft/vscode-arduino/issues/1098

sukesh-ak, 2020-09-19:
Is there someone looking into this bug?

cdelcristo, 2020-09-20:
Same Issue here

simonl65, 2020-09-21:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```

simonl65, 2020-09-21:
Related issues: #922 #1105 #1104 #1101 #1099 #1098

alandsidel, 2020-09-21:
> As a workaround:
> 
> 1. open the folder where your .ino file resides
> 2. open the .vscode folder
> 3. edit arduino.json
> 4. change the port
>    e.g.
> 
> ```
> {
>   "port":"COM4"
> }
> ```

What is this a workaround for?  It's not solving the issue mentioned here; our JSON files already have the port entry in them.

adiazulay, 2020-09-21:
> > As a workaround:
> > 
> > 1. open the folder where your .ino file resides
> > 2. open the .vscode folder
> > 3. edit arduino.json
> > 4. change the port
> >    e.g.
> > 
> > ```
> > {
> >   "port":"COM4"
> > }
> > ```
> 
> What is this a workaround for? It's not solving the issue mentioned here; our JSON files already have the port entry in them.

This workaround lets you change the port without having to use the UI.

alandsidel, 2020-09-21:
> This workaround lets you change the port without having to use the UI.

Ah.. well not being able to change the port isn't as big a problem as not being able to connect to whichever one is selected.  The Node.js version error comes up when you click the little plug to open the serial console window also.

simonl65, 2020-09-21:
@alandsidel Agreed - I'd not spotted the second issue in this thread. I was providing a workaround to the OP issue of not being able to select the serial port.

brunosacom, 2020-09-22:
I update VSCode to 1.49.1; same problem

pjgpetecodes, 2020-09-22:
Same issue here and following...

Version: 1.49.1 (user setup)
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-16T23:27:51.792Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

Arduino Extension v0.3.2.

Changing the serial port in the arduino.json fixes uploading... But the Serial Monitor is also broken...

Andrei-Florian, 2020-09-23:
Same issue here, is someone looking into this?

simonl65, 2020-09-25:
> 
> 
> Same issue here, is someone looking into this?

I believe @robotdad is looking into it (according to another related issue) - seems to be something to do with the latest Electron.

sukesh-ak, 2020-09-25:
Downgrade to 1.48.x of VS to workaround it.

fidelbrea, 2020-10-01:
+1 @sukesh-ak 
Downgrade. This is a solution. I fixed it by downgrading up to 1.48.2.
I hope you find this useful to fix your own situations.

Andrei-Florian, 2020-10-01:
Hi, sorry for the late reply. Yes the downgrade worked. Thanks for the great help!
Andrei.

kasumiru, 2020-10-18:
I update VSCode to 1.50.1; same problem

MohammedNoureldin, 2020-10-20:
@kasumiru yup, same here with 1.50.1, I cannot select the serial port..

kseyhan, 2020-10-22:
check the solution proposed here https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-714255377 if have a proper build environment on your machine.

neefhond, 2020-10-25:
Same here. Version 1.50 on Mac OS Catalina. Not resolved yet

MohammedNoureldin, 2020-10-26:
Still the same here...

escaner, 2020-10-26:
Same problem on Linux with vscode 1.50.1 and Arduino 0.3.2 extension.

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

brunosacom, 2020-10-29:
0.3.3-rc tested. 100% ok!

Version: 1.50.1 (user setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19042

MohammedNoureldin, 2020-10-29:
@brunosacom did you uninstall the extension from VS Code before installing it? I could not install what robotdad provided, because when I run the extension I get some weird error. However, I only tried it once and quickly.

robotdad, 2020-10-30:
@MohammedNoureldin that is interesting, from the error it appears it didn't like the "-rc" in the version info. I'm not sure why though. I have pushed the release now so you should be able to get that via the extension marketplace in Code now. Please raise a new issue if you still experience problems.

robotdad, 2020-10-30:
Resolved in release 0.3.3

closed 2020-10-30 by robotdad

MohammedNoureldin, 2020-10-30:
I confirm, this works now!

robotdad, 2020-10-30:
Thanks for letting me know @MohammedNoureldin

T0mHutter, 2021-02-06:
@robodad hello i just reinstalled vscode and all the extensions wanted to clean it. 
Now i can not select a Serial Port although i installed your fix. It Always jumps between Arduino path and the Serial Port 
when i manually set the port it does not work at all.
sry for my bad explaining im very bad at this

Tom

T0mHutter, 2021-02-10:
Found this and it solved my problem 
https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.5

T0mHutter, 2021-02-10:
just install it from VISX
and you're got to go

Ogieeeeee, 2021-02-10:
> 
> 
> just install it from VISX
> and you're good to go

Hey! Thank you for your fast respone! :)  
Sadly I am getting the following error and in VSCode. I can not find the latest extension version either :/ 
[image]

T0mHutter, 2021-02-10:
Did you install it from the link I sent

T0mHutter, 2021-02-10:
maybe uninstall the Arduino Extension then reinstall and then install the vsix version

Ogieeeeee, 2021-02-10:
> 
> 
> Did you install it from the link I sent
> Maybe uninstall the Arduino Extension then reinstall and then install the .vsix version

Yes, I downloaded and try'd to install the .vsix. Also try'd to uninstall and then install the extension. It says that I have to  try to install the extension in VSCode but this version (0.3.5) is not available (yet?)

T0mHutter, 2021-02-10:
yea its a user made one i guess let me see for me it worked fine and i did not get the error.

T0mHutter, 2021-02-10:
Look at this https://code.visualstudio.com/docs/editor/extension-gallery#:~:text=Install%20from%20a%20VSIX%23&text=Using%20the%20Install%20from%20VSIX,vsix%20file. 

maybe you try install it manually

alandsidel, 2021-02-10:
@T0mHutter and @Ogieeeeee can you please take your discussion private, out of the ticket?  It's not relevant to the rest of us.  Thanks!

Ogieeeeee, 2021-02-10:
@T0mHutter  I've installed it manually via the command pallet and now it works! Thank you very much!

T0mHutter, 2021-02-10:
ok but i think its helpful for people who search for the issue 
we'll speak privat from now on

Ogieeeeee, 2021-02-10:
> 
> 
> @T0mHutter I've installed it manually via the command pallet and now it works! Thank you very much! @alandsidel yea, sure.

yaksplat, 2021-02-13:
> Look at this https://code.visualstudio.com/docs/editor/extension-gallery#:~:text=Install%20from%20a%20VSIX%23&text=Using%20the%20Install%20from%20VSIX,vsix%20file.
> 
> maybe you try install it manually
That fixed it, thank you

khorbushko, 2021-04-14:
Looks like the issue is can be reproduced on new M1 from Apple - can't select neither port neither open serial port (if manually write it in config). Upload works fine. 

When clicked on "open port"button
[image]

adiazulay, 2021-04-14:
Can you try updating the extension version 0.4.0? 
That might resolve this, but I think this more likely is an issue with the M1, since I don't think the native modules we use are compiled against ARM architecture yet.

khorbushko, 2021-04-14:
I tested all starting from 0.3.0 to 0.4.0 .

do u know maybe it's planned to work on support for M1?

[image]

msm52, 2021-05-02:
Same in my M1

[19 more replies]