microsoft/vscode-arduino #1100: Verify/Upload uses wrong file as "main file"

briman0094, 2020-09-13:
I'm starting an Arduino project that will span multiple boards. I have subfolders for each board's project, such that the filesystem structure looks like this:

```
.vscode/
  |- arduino.json
  |- c_cpp_properties.json
  |- settings.json
board1/
  |- main_board1.ino
board2/
  |- main_board2.ino
  |- Utility.h
obj/    # This is the "output" folder in arduino.json
```

This seems to work fine at first, despite the inconvenience of having to manually switch which `.ino` sketch is active in `arduino.json`. However, I discovered once I added that `Utility.h` file in `board2/` that it seems the Upload/Verify commands are passing the wrong "main" file to Arduino when there is more than one file in the folder.

`main_board1.ino` does not include anything, and `main_board2.ino` starts with this:

```cpp
#include "Utility.h"
```

When `board1/main_board1.ino` is the active sketch, it verifies fine. However when `board2/main_board2.ino` is the active sketch, it throws a compile error:

```
main_board2:1:10: fatal error: Utility.h: No such file or directory

 #include "Utility.h"

          ^~~~~~~~~~~
```

Oddly, if I create a `Utility.cpp` file and have _it_ include `Utility.h`, it compiles fine. It doesn't have to have anything in it other than `#include "Utility.h"`.

I looked at the verbose build logs and noticed something. When compiling `board1/main_board1.ino` I see the following:

```
D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\[Username]\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\[Username]\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Users\[Username]\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10813 -build-path c:\Development\Projects\Arduino\BugReproduction\obj -warnings=null -prefs=build.path=c:\Development\Projects\Arduino\BugReproduction\obj -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Arduino\hardware\tools\avr -verbose c:\Development\Projects\Arduino\BugReproduction\board1\main_board1.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\[Username]\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\[Username]\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Users\[Username]\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10813 -build-path c:\Development\Projects\Arduino\BugReproduction\obj -warnings=null -prefs=build.path=c:\Development\Projects\Arduino\BugReproduction\obj -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Arduino\hardware\tools\avr -verbose c:\Development\Projects\Arduino\BugReproduction\board1\main_board1.ino
```

However when compiling `board2/main_board2.ino`, I see the following:

```
D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\[Username]\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\[Username]\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Users\[Username]\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10813 -build-path c:\Development\Projects\Arduino\BugReproduction\obj -warnings=null -prefs=build.path=c:\Development\Projects\Arduino\BugReproduction\obj -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Arduino\hardware\tools\avr -verbose c:\Development\Projects\Arduino\BugReproduction\board2\Utility.h
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\[Username]\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\[Username]\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Users\[Username]\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10813 -build-path c:\Development\Projects\Arduino\BugReproduction\obj -warnings=null -prefs=build.path=c:\Development\Projects\Arduino\BugReproduction\obj -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=D:\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Arduino\hardware\tools\avr -verbose c:\Development\Projects\Arduino\BugReproduction\board2\Utility.h
```

The thing to note is that when compiling `board1/main_board1.ino`, the "entry point" file passed to `arduino-build` is in fact `board1/main_board1.ino`, yet when compiling `board2/main_board2.ino`, the "entry point" is actually `Utility.h`. It tries to compile a `Utility.cpp` file even if one does not exist.

In the `obj/sketch/` folder, I can see files named `Utility.cpp.cpp`, `Utility.cpp.cpp.d`, and `Utility.cpp.cpp.o`.

If I manually invoke the above commands for `board2/main_board2.ino` but change the final argument to `arduino-build` from the `Utility.h` file to the `board2/main_board2.ino` file, it compiles fine even without the `Utility.cpp` file present.

I've attached a small reproduction project:
[BugReproduction.zip](https://github.com/microsoft/vscode-arduino/files/5215028/BugReproduction.zip)

briman0094, 2020-09-13:
I noticed when attempting to call `arduino_debug.exe` from the command line without the `--verify` flag, it complains that `main_board2.ino` should instead be named `board2.ino`. Apparently it requires the sketch name to match the folder name (for some apparently arbitrary reason). Renaming it to `board2.ino` allows it to compile fine without `Utility.cpp` present. It seems this may actually be an underlying bug in the Arduino IDE; it may be prudent for `vscode-arduino` to warn about the `.ino` file being named wrong unless Arduino acknowledges this as a bug and fixes it.

koderian, 2020-09-23:
I have the same problem, when verifying/uploading, the compiler sometimes verifies all my recently opened files that are open in the editor, so it's verifying like 6-7 files. I'd rather it just verify/upload the sketch that I asked it to verify/upload (and others, if my code imports from other sketches.) 

I don't know how to fix this, sometimes restarting VS code fixes it, but not. . . now. 

I know the Arduino.json has a `"sketch": "arduino\\1.ino,`, and it seems to be correctly updating, but changing the sketch filename in that json to different file doesn't make it verify/upload the sketch I want to.

fabquenneville, 2021-01-29:
Hello,

Started coding with arduino on vscode and I also ran into this issue where I have to delete the sketch line in arduino.json before uploading from a multi-projects workplace.

I would +1 on the feature request above