microsoft/vscode-arduino #1101: Cant select Serial Port

khiemnt00, 2020-09-16:
Issue Type: <b>Bug</b>

In version 1.49.0, i cant not able to select serial port

Extension version: 0.3.2
VS Code version: Code 1.49.0 (e790b931385d72cf5669fcefc51cdf65990efa5d, 2020-09-10T13:22:08.892Z)
OS version: Windows_NT x64 10.0.19041

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Xeon(R) CPU E3-1230 v5 @ 3.40GHz (8 x 3408)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.95GB (9.09GB free)|
|Process Argv|. --crash-reporter-id 2f0618bb-7cd7-4f6f-959c-ab1eb4c18e74|
|Screen Reader|no|
|VM|0%|
</details>

maxulm, 2020-09-18:
#1099 Duplicate

simonl65, 2020-09-21:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```

simonl65, 2020-09-21:
Related issues: #922 #1105 #1104 #1101 #1099 #1098

fidelbrea, 2020-10-01:
Downgrade. This is a solution. I fixed it by downgrading up to 1.48.2.
I hope you find this useful to fix your own situations.

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

robotdad, 2020-10-30:
Resolved in release 0.3.3

closed 2020-10-30 by robotdad