microsoft/vscode-arduino #1103: Working inside of workspaces

DeadlySurgeon, 2020-09-17:
My general workflow (with other languages, first time using this extension) has me leveraging Workspaces so it is easier to reference other parts of my code. This means multiple, different folders open at once. As a result it might look something similar to this:

[image]

When I go to run the command `Arduino: Verify`, or any other command for that matter, the extension will run the command against the entrypoint in the first project opposed to the entrypoint of the folder belonging to the currently opened file. I'm not entirely sure if this is a bug, but it prevents this extension from being used in conjunction with Workspaces. The workaround is to just have one window per project, but it would be nice if this was fixed.

On another, offtopic note: Is this resource dead or something? There are several critical bugs that seem to have gone unaddressed for quite some time.

MonWithNoName, 2020-09-29:
Man it would be heavenly if they would add this. As a newer programmer just dipping my toes into the hobby, having to make completely new folders for each project is a nightmare. I would also like to know if the extension is dead, been waiting since august and still no fix to the serial port bug to be found.

DeadlySurgeon, 2020-09-30:
So, oddly enough the serial port bug just randomly stopped for me. No idea how or why.

HEMISixty, 2020-09-30:
Do you mean the serial port is now functioning?  If so, what is your electron version, please?

DeadlySurgeon, 2020-10-05:
I'll get it to you after work, and I mean, it _seemed_ to work for me?

DeadlySurgeon, 2020-10-05:
I think that I had to manually write the config for the serial port location.

HEMISixty, 2020-10-13:
Oh, ok, no worries...

DeadlySurgeon, 2020-10-13:
@HEMISixty 
So yeah it turns out I just had to manually specify the port.

``` json
{
  "sketch": "app.ino",
  "board": "esp32:esp32:esp32wrover",
  "configuration": "PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none",
  "port": "/dev/cu.SLAB_USBtoUART"
}
```

threeal, 2021-01-19:
can't wait for someone to implement this. either by choosing which file to be used in the option first or compile the file that currently being opened. the latter method seems to be the best choice, but it would be a problem if someone is opening two different file.

geepy, 2021-02-14:
I would greatkly appreciate having different settings for different .ino files in one workspace. I have aproject combining Uno for measuring, NodeMCU for transmitting data and a pure ESP12E as MQTT broker. Having to fuddle with the config file every time I work on another part of the solution is a PITA...